openwrt/luci #7410: RFC: update default setting for odhcp6c "norelease"

systemcrash, 2024-11-19:
Recent RFC have updated recommendations surrounding this behaviour to `SHOULD NOT`, so I'm open to wisdom surrounding such a change. 

See https://www.rfc-editor.org/rfc/rfc9096.html#name-improved-customer-edge-rout
"""
WPD-9:
    CE routers SHOULD NOT automatically send DHCPv6-PD RELEASE messages upon restart events. See [Section 3.1](https://www.rfc-editor.org/rfc/rfc9096.html#dhcpv6-release) for further details.
"""

and https://www.rfc-editor.org/rfc/rfc9096.html#name-automatic-dhcpv6-releases
"""
[3.1. ](https://www.rfc-editor.org/rfc/rfc9096.html#section-3.1)[Automatic DHCPv6 RELEASEs](https://www.rfc-editor.org/rfc/rfc9096.html#name-automatic-dhcpv6-releases)

Some CE routers are known to automatically send DHCPv6-PD RELEASE messages upon restart events. However, this may inadvertently trigger a flash-renumbering scenario, along with the associated problems discussed in [[RFC8978](https://www.rfc-editor.org/rfc/rfc9096.html#RFC8978)], that this document attempts to mitigate.
As a result, requirement WPD-9 from [Section 3](https://www.rfc-editor.org/rfc/rfc9096.html#CPE) specifies that CE routers SHOULD NOT automatically send DHCPv6-PD RELEASE messages upon restart events.
"""

@jow- 
@dannil 
@hnyman 
@feckert 
@ynezz

systemcrash, 2024-11-19:
https://github.com/openwrt/luci/blob/b9496f2023cb6b5f51dff88f1cf3a3063390dae4/protocols/luci-proto-ipv6/htdocs/luci-static/resources/protocol/dhcpv6.js#L32-L33

Set default to 1 (enable norelease)

dannil, 2024-11-29:
I'm not well-versed on how DHCPv6 works under the hood, but if the updated recommendation regarding how DHCPv6-PD RELEASE messages should be handled has changed it seems like a reasonable change.

systemcrash, 2024-11-30:
Closed by 98c04137b47dd5253cdfcf18e5b4ad6e68258c46.

closed 2024-11-30 by systemcrash