openwrt/luci #7412: 23.05.5/packages/aarch64_cortex-a72/luci/luci-app-attendedsysupgrade_git-24.319.71435-cd1e372_all.ipk returns 404 error

masterofcodepc, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

Several packages in https://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a72/luci/ are returning 404 errors.
For example: https://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a72/luci/luci-app-attendedsysupgrade_git-24.319.71435-cd1e372_all.ipk

This is preventing the installation of packages on 23.05.5 on raspberry pi 4.

[image]
[image]

I've tried using opkg to install luci-app-attendedsysupgrade, and get errors, I tested using wget which gets 404 errors and I have go to the web packages list in a browser and clicked on the the package name from the list and it returns a 404 error. It looks to be an issue in the 23.05.5/packages/aarch64_cortex-a72/luci/ folder as many packages are also returning 404 errors, but other architectures are not returning 404 for the same packages.

### Actual behaviour

On a raspberry pi 4 running 23.05.5 I tried to install luci-app-attendedsysupgrade from the Luci interface, it failed. So I tried the opk that also failed so I tried using wget to pull the package from the repo and it gets a 404 error. I then tried going to the packages site in a browser and navigating to the package, it also returns 404. 
I found that several packages in the releases/23.05.5/packages/aarch64_cortex-a72/luci/ folder are returning 404 errors.

### Expected behaviour

Installing a package from the luci interface would not have errors.

### Steps to reproduce

on a RPi 4 running 23.05.5 (64 bit) try to install luci-app-attendedsysupgrade.
several other packages are also returning 404 errors, but this one is the specific one i was looking for.

### Additional Information

```shell
?? Im not sure how this applies in my case.
```

### Relevant log output

```shell
Firefox
```

systemcrash, 2024-11-21:
Looks like a transient error whereby you probably just needed to run an update first.

closed 2024-11-21 by systemcrash

hnyman, 2024-11-23:
Currently the directory page shows
`luci-app-attendedsysupgrade_git-24.322.79771-ee4c08d_all.ipk	5.5 KB	Wed Nov 20 23:48:58 2024
`
So the version has changed and the old package is not there any more.

You have managed to try just when buildbot has been updating the packages.
(The release packages are not static, but are rebuilt if there are changes into them in the 23.05 branch)

masterofcodepc, 2024-11-23:
Thank you, I see. It wasn't that way on the 19th when I ran int the issue oir the 20th when I  when I submitted the ticket, but it has updated now so I am good. 

Thanks!