openwrt/luci #7413: luci-bwc: should not use /var/lib/luci-bwc but /tmp/luci-bwc

M95D, 2024-11-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

Luci-bwc stores data in /var/lib/luci-bwc. Some users prefer a permanent /var. That's usually on a USB stick or router's flash. Writing there increases wear.

### Expected behaviour

Luci-bwc should write to /tmp/luci-bwc instead.

### Steps to reproduce

No need.

### Additional Information

```shell
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28116+17-cd92cbddf8"
OPENWRT_BOARD="mvebu/cortexa9"
OPENWRT_ARCH="arm_cortex-a9_vfpv3"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28116+17-cd92cbddf8"
OPENWRT_BUILD_DATE="1731864854"
```

systemcrash, 2024-11-21:
Reasonable, although it won't survive reboots. @jow- ?

M95D, 2024-11-21:
What reason is there for it to survive reboots? In fact, why not clear it when luci-bwc exits?

jow-, 2024-12-04:
https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch05.html#purpose31

> /var contains variable data files. This includes spool directories and files, administrative and logging data, and transient and temporary files.

closed 2024-12-04 by M95D