openwrt/luci #7418: Feature request: Add fortinet to openconnect protocol for 23.05

n0la, 2024-11-22:
### What would you like to see in luci?

luci in 23.05 does not support fortinet as a `vpn_protocol` for openconnect, even though openconnect does support it. So you have a disconnect were you can conf "fortinet" as a vpn protocol with uci, but luci won't show it properly:

```
network.workvpn=interface
network.workvpn.proto='openconnect'
network.workvpn.vpn_protocol='fortinet'
```

This VPN works just fine for me on my router, but luci won't show that correctly (see the "CISCO AnyConnect" compared to "Fortinet"):

[image]

Would it be possible to just add "fortinet" as a possible protocol option for luci?

systemcrash, 2024-11-22:
Fixes for this came in a while back but weren't cherry picked to 23. 

It would seem we can remove the openfortinet protocol and package entirely if the openconnect suite does everything we need. 

Are there any missing properties or elements specific to fortinet in the openconnect proto?

systemcrash, 2024-11-22:
That should be fixed now by 5a09e672a9c5c42d370571dada283ae2dfbc1247.

closed 2024-11-22 by systemcrash

n0la, 2024-11-28:
Thanks, I updated my LUCI recently on my OpenWRT and I can now conf the openforti connection properly!

systemcrash, 2024-11-28:
Great to hear - thanks for confirmation!