openwrt/luci #7420: Feature request: luci-app-wireguard save peer AllowedIPs configuration

brianjmurrell, 2024-11-24:
### What would you like to see in luci?

When generating a QR code for a peer, there is a field to define the `AllowedIPs` that the generated configuration/QR code should instruct the client to configure.  This value defaults to `0.0.0.0/0, ::/0` which results in all traffic being routed through the VPN, which may not be what is desired.

But this value can be edited on that page to change the configuration that the QR code describes but that change is only valid while that page is open and needs to be adjusted every time one enters that page to generate a QR.

It would be nice if there was a way to persist this preference in the config in some way so that the generated QR is correct per the admin's desired value for a client's `AllowedIPs` without having to make that adjustment every time a QR code generation is requested.

systemcrash, 2024-11-24:
This already exists in the edit peer dialogue. They're called.... Allowed IPs, followed by Route Allowed IPs checkbox. Open that dialogue and context changes to the point of view of the remote peer: we have a new Allowed IPs in this qrcode dialogue to ensure the client who will receive this tunnel route all their traffic over it. Suitable for "most situations". 

Return to the edit peer dialogue and the context, or point of view, is back to that of the router you're configuring. It doesn't need 0.0.0.0 to reach the peer - it likely only needs a single IP.

How can this be improved? (It can, it just requires a bit of thought)

brianjmurrell, 2024-11-24:
Your description is that of the peer's entry on the OpenWrt server.  The AllowedIPs on the page you are describing turns into the `list allowed_ips` in `/etc/config/network`.

As I said previously, I am referring the QR code that is generated when one clicks on _Configuration Export_ on that same page.  It's the _Allowed IPs_ in **that** page which the user needs to adjust and which adjustments to not persist, so they need [re-]adjusting **every time** you go into that page.  Those should be persisting somehow/where IMO.

Captain-Nemo-Xu, 2026-09-24:
I have the same issue on ImmortalWrt 25.12.1 with `luci-proto-wireguard` 26.236.50544~cb5d434.

The problem is not only that `AllowedIPs` in the "Generate configuration" dialog is not persisted. The generated client configuration also needs persistent server-side defaults for values that are common to all peers.

My setup is a typical WireGuard server:

* WireGuard server: `10.66.66.1/24`
* Server listening port: `51820/UDP`
* Public endpoint: a DDNS hostname, e.g. `vpn.example.com:51820`
* LAN network: `192.168.8.0/24`
* Multiple clients:

  * Client 1: `10.66.66.2/32`
  * Client 2: `10.66.66.3/32`
  * Client 3: `10.66.66.4/32`

For every client, the generated configuration should normally contain the same values:

```ini
[Peer]
PublicKey = <server public key>
Endpoint = vpn.example.com:51820
AllowedIPs = 192.168.8.0/24
```

Only the client's own tunnel address and keys are different.

I checked the actual `wireguard.js` shipped with my system:

`/www/luci-static/resources/protocol/wireguard.js`

The current implementation contains:

```js
const ips=['0.0.0.0/0','::/0'];
```

and creates the configuration generator with:

```js
config:{
    endpoint:hostnames[0],
    allowed_ips:ips,
    addresses:eips,
    dns_servers:dns
}
```

Therefore `AllowedIPs` is currently hard-coded as a temporary default in the configuration generator rather than being stored as a persistent server-side setting.

The `Endpoint` handling is also problematic for setups where DDNS is managed outside OpenWrt's UCI `ddns` system. In my case, DDNS is managed by Lucky, so LuCI cannot obtain the hostname through:

```js
uci.load('ddns')
```

As a result, I have to manually enter the same endpoint every time I generate a new peer configuration.

### Suggested improvement

It would be useful to provide persistent defaults for the client configuration generator at the WireGuard interface/server level, for example:

```text
Client configuration defaults
    Endpoint: vpn.example.com:51820
    Allowed IPs: 192.168.8.0/24
    DNS: ...
```

These values should be used as defaults whenever "Generate configuration" is opened for any peer, while still allowing them to be overridden for an individual export.

The important distinction is:

* Peer `AllowedIPs` in `/etc/config/network` identifies the peer on the server, e.g. `10.66.66.2/32`.
* Client-side `AllowedIPs` determines which destination networks the client routes through the tunnel, e.g. `192.168.8.0/24`.
* `Endpoint` is the address that the generated client uses to connect to the WireGuard server.

These are different configuration values and should not be conflated.

This would make the configuration generator much more practical for a WireGuard server with multiple clients, especially when all clients share the same endpoint and routed networks.