openwrt/luci #7423: luci-app-ddns: Toggle Button Won't Work.

Try2079, 2024-11-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

Hit "Stop DDNS" button, nothing happened.

CLI: logread
nothing new

### Expected behaviour

Toggle DDNS service status.

### Steps to reproduce

Hit "Stop DDNS" button.
--- ---

And source code snippet below:
`/luci-app-ddns/htdocs/luci-static/resources/view/ddns/overview.js`
`line 169~175`
```javascript
1. handleToggleDDns: function(m, ev) {
2. 	return this.callInitAction('ddns', 'enabled')
3. 		.then(L.bind(function (action) { return this.callInitAction('ddns', action ? 'disable' : 'enable')}, this))
4. 		.then(L.bind(function (action) { return this.callInitAction('ddns', action ? 'stop' : 'start')}, this))
5. 		.then(L.bind(m.render, m))
6. 		.catch(function(e) { ui.addNotification(null, E('p', e.message)) });
7. },
```
line 2 (line 170): When I run `ubus call luci setInitAction "{'name':'ddns', 'action':'enabled'}"`, 
it return an error:
`{ "error": "Invalid action" }`
==**Seems like no such an action as 'enabled'.**==

### Additional Information

```shell
openwrt version = 23.05.5
```

systemcrash, 2024-11-25:
`enabled` is not an action, it's a state. `enable`, however, is an action. 
https://github.com/openwrt/luci/blob/2ce85291bdd70accbb1863bfb32ddf18bcb7b3fe/modules/luci-base/root/usr/share/rpcd/ucode/luci#L46-L67

Try2079, 2024-11-26:
Submitted a [pull request #7425](https://github.com/openwrt/luci/pull/7425). I'm not very good at using github, but I really want to make my contribution. Please forgive me if there are any mistakes.