openwrt/luci #7428: luci: custom firewall rules don't show up with fw4

YoavNahumus, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

In the network>firewall tab of the GUI, the custom rules tab does not show up
I was able to fix this by removing the dependency of /usr/share/fw3/helpers.conf from the
/usr/share/luci/menu.d/luci-app-firewall.json file

### Expected behaviour

The custom tab in the firewall will show up and allow the user to put nftables commands for rules

### Steps to reproduce

go to: Network -> Firewall and look at the tabs on a device using firewall4

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.2"
VERSION_ID="23.05.2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r23630-842932a63d"
OPENWRT_BOARD="ipq40xx/generic"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.2 r23630-842932a63d"
```

YoavNahumus, 2024-11-26:
> Why not start with latest stable, 23.05.5, to make sure it's not fixed?

the luci-app-firewall.json still has the dependency there

systemcrash, 2024-11-27:
> ###
> 
> In the network>firewall tab of the GUI, the custom rules tab does not show up I was able to fix this by removing the dependency of /usr/share/fw3/helpers.conf from the /usr/share/luci/menu.d/luci-app-firewall.json file

One of the many idiosyncrasies in the system: it's a hold-over from fw3 days. Not everyone runs fw4; some still use iptables, and the custom rules window is meant for fw3 and iptables commands.

Those are written to `/etc/firewall.user`. 

https://github.com/openwrt/firewall4/blob/dfbcc1cd127c78fc61bb870d36d2512b571d223b/root/usr/share/ucode/fw4.uc#L3210-L3212

https://github.com/openwrt/firewall4/blob/dfbcc1cd127c78fc61bb870d36d2512b571d223b/tests/06_includes/02_firewall.user_include#L42-L57

brada4, 2024-11-30:
You can add inserts to rules in `/etc/nftables.d/*.nft`
They go in
```
table inet fw4 {
   ... setup interfaces variables....
   ... offload if present ...
   chain yourchain {
     hook xxx prio yyy;
      iifname lo counter
   }

Or in /usr/share in other places.

systemcrash, 2024-11-30:
Perhaps we can parse that folder content as an alternative when the user runs fw4.
@stokito @dannil @Ramon00 any takers for that?

systemcrash, 2024-11-30:
`/usr/sbin/nft --json`?

systemcrash, 2024-11-30:
That's basically what the firewall status page uses so I don’t see why not. Although here the idea was actual file content. What did you have in mind?

Ramon00, 2024-12-08:
Is it just an matter of removing the depend?
`	"admin/network/firewall/custom": {
		"title": "Custom Rules",
		"order": 50,
		"action": {
			"type": "view",
			"path": "firewall/custom"
		},
		"depends": {
			"fs": { "/usr/share/fw3/helpers.conf": "file" }
		}
	}`

skleeschulte, 2025-01-17:
> One of the many idiosyncrasies in the system: it's a hold-over from fw3 days. Not everyone runs fw4; some still use iptables, and the custom rules window is meant for fw3 and iptables commands.
> 
> Those are written to `/etc/firewall.user`.

The wiki says it a bit different: _Custom rule inclusion through a shell script works similarly as fw3, but the script should use nftables._ (See [Config include section with shell script](https://openwrt.org/docs/guide-user/firewall/firewall_configuration?rev=1732451115#includes_2203_and_later_with_fw4))

Adding rules with shell commands can be a quick and easy way to test rules, e.g. just copy some commands from a tutorial and restart the firewall.

My use case is another one: On OpenWrt 23.05.5 with fw4 I installed packages `iptables-nft` and `iptables-mod-ipopt` to be able to increase packages' ttl, which apparently cannot be done with nftables. I created /etc/firewall.user, chmod it to 0600 and added to the file:

```
iptables -t mangle -F
iptables -t mangle -A PREROUTING -i br-lan -d 239.255.255.250 -j TTL --ttl-inc 1

```
Works like a charm, and would be handy to be able to do it from the GUI.

brada4, 2025-01-17:
ttl-inc option is not ported to iptables-nft. 
```
iptables-translate -t mangle -A PREROUTING -i br-lan -d 239.255.255.250 -j TTL --ttl-inc 1
iptables-translate v1.8.10 (nf_tables): unknown option "--ttl-inc"
Try `iptables-translate -h' or 'iptables-translate --help' for more information.
```

brada4, 2025-01-17:
You have to use set ttl probably with  
```
map ttlinc ttl : ttl {
  1:2 .2:3 .... 254:255
}
```
(not guaranteed to work, jut meta-guess that it coudl)

skleeschulte, 2025-01-18:
> ttl-inc option is not ported to iptables-nft.
> 
> ```
> iptables-translate -t mangle -A PREROUTING -i br-lan -d 239.255.255.250 -j TTL --ttl-inc 1
> iptables-translate v1.8.10 (nf_tables): unknown option "--ttl-inc"
> Try `iptables-translate -h' or 'iptables-translate --help' for more information.
> ```

On my system, the output looks different:
```
root@openwrt:~# iptables-translate -t mangle -A PREROUTING -i br-lan -d 239.255.255.250 -j TTL --ttl-inc 1
nft # -t mangle -A PREROUTING -i br-lan -d 239.255.255.250 -j TTL --ttl-inc 1
```
The commands in /etc/firewall.user listed in my [post above](https://github.com/openwrt/luci/issues/7428#issuecomment-2598437486) also work well - an xt-ttl-rule is added to nftables and the ttl-inc rule is added to iptables. Maybe the package `iptables-mod-ipopt` is missing in your test-environment?

But my main point was: there are use cases for using command line firewall rules with fw4/nftables, and thus it would still be nice to have an easy way to edit /etc/firewall.user.

brada4, 2025-01-18:
I have xt bridge blacklisted intentionally.