openwrt/luci #7432: luci-app-banip: Feed selection is broken

vincejv, 2024-11-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

Maintainer: @dibdot
[image]

### Actual behaviour

Doesn't let me enable and disable the block lists provided by banIP from Luci

### Expected behaviour

Feed selection from Luci is available

### Steps to reproduce

1. Navigate to Luci > Services > banIP
2. Go to feed selection tab

### Additional Information

```shell
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28214-5a4eb56a7b"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28214-5a4eb56a7b"
OPENWRT_BUILD_DATE="1732829601"
```

### What browsers do you see the problem on?

Firefox, Chrome, Safari, Microsoft Edge

dibdot, 2024-11-30:
Not reproducible, just re-tested with latest master ...
[image]

Most probably the JSON-files of your base package are (for whatever reason) borked. Just re-install the latest banIP base-package.

vincejv, 2024-11-30:
@dibdot is af611bce44 a self-built image? I reproduced it too with other platforms ramips example, this is from clean install btw
[image]

vincejv, 2024-11-30:
can anyone else confirm/ help repro?

dibdot, 2024-11-30:
Please provide the output of `ls -la /etc/banip/` and of `cat /etc/banip/banip.custom.feeds` - thanks.

vincejv, 2024-11-30:
`root@EW1200G:~# ls -la /etc/banip`
```
drwxr-xr-x    2 root     root             0 Dec  1 02:12 .
drwxr-xr-x    1 root     root             0 Dec  1 02:12 ..
-rw-------    1 root     root             0 Nov 29 19:03 banip.allowlist
-rw-------    1 root     root             0 Nov 29 19:03 banip.blocklist
-rw-------    1 root     root          4915 Nov 29 19:03 banip.countries
-rw-------    1 root     root             0 Nov 29 19:03 banip.custom.feeds
-rw-------    1 root     root         18388 Nov 29 19:03 banip.feeds
-rw-------    1 root     root          1798 Nov 29 19:03 banip.tpl
```

second command prints blank
```
root@EW1200G:~# cat /etc/banip/banip.custom.feeds
root@EW1200G:~# 
```

I think i noticed the issue when luci base was upgraded..

vincejv, 2024-11-30:
sorry for the trouble, seems i like i have figured out the issue, something wrong with my uhttpd configuration.

closed 2024-11-30 by vincejv

systemcrash, 2024-11-30:
What might that be? Perhaps a fix is actually needed elsewhere?

kmau, 2024-12-26:
Hi @vincejv, I have the same issue. From PC (Chrome & Firefox) is broken like in your screenshot, from phone (iPhone/Safari) is fine.
How did you fix your uhttpd config?

kmau, 2025-01-10:
Hi @dibdot, since I also having the issue, can you reopen the issue?
Could I also send ls -la /etc/banip/ and of cat /etc/banip/banip.custom.feeds and, if you kindly point me how, my uhttpd configuration, since @vincejv fixed there?
Thank you in advance.

vincejv, 2025-01-10:
Hello @kmau yes this has something to do with the webserver/proxy, try directly connecting to luci without proxy or anything in between

kmau, 2025-01-10:
Hi @vincejv yes, as I stated in the previous comment (beetween the many edits), if I go with the IP address of the router it's work like a charm :) 
Thanks anyway

vincejv, 2025-03-22:
my remedy/solution was to change HTTP Proxy from 1.1 to 1.0, example in nginx

```
proxy_http_version 1.0
```

webysther, 2025-03-24:
> my remedy/solution was to change HTTP Proxy from 1.1 to 1.0, example in nginx
> 
> ```
> proxy_http_version 1.0
> ```

Oh! Disable websockets support did the trick because this option do:

```conf
proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection $http_connection;
proxy_http_version 1.1;
```

But I have another custom conf and I need to add the 1.0 on the location, works like a charm, thanks!

webysther, 2025-03-24:
This is a bug for me anyway, banip forcing downgrade the HTTP version is bad

webysther, 2025-03-26:
> my remedy/solution was to change HTTP Proxy from 1.1 to 1.0, example in nginx
> 
> ```
> proxy_http_version 1.0
> ```

To reduce the surface of the downgrade I did this and works:

```nginx
location /cgi-bin/cgi-download {
    proxy_http_version 1.0;
}
```