openwrt/luci #7436: <luci-proto-wireguard>: The peer name detection fails when the endpoint_host is an IPv6 address.

Stream-z, 2024-12-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

When the endpoint_host is set to an IPv6 address, the peer name appears as empty on the Status → WireGuard menu. When set to an IPv4 address, the peer name is displayed correctly.

### Expected behaviour

1. When the endpoint_host is set to an IPv6 address,the peer name should be displayed correctly on the Status → WireGuard page.

### Steps to reproduce

1.go to:Network → Interfaces → wireguard → peer tab
2.modify the peer's endpoint_host to an IPv6 address or a domain name that resolves to an IPv6 address.

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.5"
VERSION_ID="23.05.5"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24106-10cc5fcd00"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.5 r24106-10cc5fcd00"
```

### What browsers do you see the problem on?

Chrome

Stream-z, 2024-12-01:
I am not very familiar with this area of knowledge, so I tried using AI tools to locate the issue. I found that the problem might be in the ``checkPeerHost`` function in the ``luci.wireguard`` file:
```javascript
function checkPeerHost(configHost, configPort, wgHost) { 
    const ips = popen(`resolveip ${configHost} 2>/dev/null`);
    if (ips) {
        for (let line = ips.read('line'); length(line); line = ips.read('line')) {
            const ip = rtrim(line, '\n');
            if (ip + ":" + configPort == wgHost) {
                return true;
            }
        }
    }
    return false;
}
```
The issue is that when ``wgHost`` is an IPv6 address, the address format includes square brackets``[]``, but the IPv6 address parsed by resolveip does not include square brackets``[]``. As a result, even if the IPv6 address is the same, the function returns false, which I believe is incorrect."

this-username-has-been-taken, 2024-12-01:
Hi!

That was my code - my apologies.
Can you please share your peer configuration from the GUI and `/etc/config/network`? I'd like to check how ipv6 is configured and looks like in the interface and config. Especially I am interested to see `option endpoint_host` value from the `/etc/config/network` file.
Please do not forget to mask/remove all the sensitive information.
I don't have an IP v6 to replicate the issue.

Stream-z, 2024-12-02:
> Hi!
> 
> That was my code - my apologies. Can you please share your peer configuration from the GUI and `/etc/config/network`? I'd like to check how ipv6 is configured and looks like in the interface and config. Especially I am interested to see `option endpoint_host` value from the `/etc/config/network` file. Please do not forget to mask/remove all the sensitive information. I don't have an IP v6 to replicate the issue.
> 
> **UPD1:** Taking into account your suggestion the fix might be:
> 
> ```
> // replacing "[" and "]" with empty string in the wgHost variable
> const hostIp = replace(wgHost, /[[\]]/g, "")
> ...
> // inside the loop
> if (ip + ":" + configPort == hostIp) {
> ```
> 
> or
> 
> ```
> // adding square brackets to IP and checking again
> if (ip + ":" + configPort == wgHost || "[" + ip + "]" + ":" + configPort == wgHost) {
> ```
> 
> **UPD2:** Can you please also share command output: `wg show all dump`?

Thank you for your reply. 
This is  my peer configuration from the  ``/etc/config/network`` :
```bash
config wireguard_wg0
        option preshared_key 'my_preshared_key'
        option route_allowed_ips '1'
        option endpoint_port 'my_endpoint_port'
        option persistent_keepalive '25'
        option description 'rd'
        option public_key 'my_public_key'
        list allowed_ips '192.168.5.0/24'
        list allowed_ips '192.168.4.3'
        option endpoint_host 'xxx.xxx.top' # resolved to an IPv6 address, or configure as an IPv6 address, e.g., 2001:0db8:0000:0000:0000:0000:0000:0001 (without square brackets, as adding square brackets will make the configuration invalid)
```

My peer configuration from the GUI:
[image]

The command output of `wg show all dump`:
[image]

I tried the first method you provided, but it seems there were some issues. It looks like replacing the `[` and `]` characters didn’t work as expected. I made some modifications to the code as follows:

```javascript
const hostIp = replace(wgHost, /\[|\]/g, "");
```
With this change, it worked correctly, and the peer name is now displayed as expected. The second method also works, and the peer name is displayed properly as well.

this-username-has-been-taken, 2024-12-02:
Thank you!

So, `const hostIp = replace(wgHost, /\[|\]/g, "");` it will be

closed 2024-12-02 by systemcrash