openwrt/luci #7448: luci-app-package-manager: Package list table header still says `Size: (.ipk)` when using APK package manager

vincejv, 2024-12-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

Might have missed the change in this PR: https://github.com/openwrt/luci/pull/7341
[image]

### Actual behaviour

Table header of package list still shows `.ipk` even apk is used

### Expected behaviour

It should show `.apk` when APK is used/detected

### Steps to reproduce

1. Go to System > Software
2. Click Update Lists (optional) to populate the table
3. Table header is shown wrong

### Additional Information

```shell
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28198-341cc047b9"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28198-341cc047b9"
OPENWRT_BUILD_DATE="1732597653"
```

### What browsers do you see the problem on?

Firefox, Chrome, Safari, Microsoft Edge

vincejv, 2024-12-06:
good day @systemcrash, thanks for the time addressing the issue
`E('th', { 'class': 'th col-1 center size'}, [ _('Size (%s)').format(L.hasSystemFeature('apk') ? '.ipk' : '.apk') ]),`

but it seems like https://github.com/openwrt/luci/commit/9a5a5ec5784848a60e3c10895d762d3a69b3d13e has the conditional statement should be the other way around.

systemcrash, 2024-12-06:
You're right 🤦

systemcrash, 2024-12-06:
Fixed in 276fef82e0095731f0b59501ca3d62059fd5fede