openwrt/luci #7451: luci-app-mwan3: Setting blocks getting mixed up if they have the same name

z411, 2024-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

If I create either an interface, member, policy or rule with the same name as any other interface, member, policy or rule, it will overwrite the previous block and mix the values up silently.

As you can see in the screenshot, I created an interface called "vtr" then created a policy with the same name, and as you can see the policy got created with the values of the policy AND the interface, and the interface got deleted.

### Expected behaviour

I'm new to OpenWRT and I'm not sure if blocks of different type but with the same name are allowed (mwan3 certainly doesn't allow them) but if it's not, an error should be thrown in LuCI if you try to create something with the same name of another thing.

### Steps to reproduce

1. Create an interface with a specific name
2. Create any other thing (member, policy, rule, etc.) with the exact same name
3. Watch the interface get deleted and the values get mixed up in the config file

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.5"
VERSION_ID="23.05.5"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24106-10cc5fcd00"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.5 r24106-10cc5fcd00"
```

### What browsers do you see the problem on?

Firefox, Chrome

jow-, 2024-12-04:
It probably should, maybe there's a bug somewhere.

systemcrash, 2024-12-06:
I 'fixed' this bug 94f187bc9b2d9a56e7400972d423e719f4940c1a in September. Upgrade your mwan3.

systemcrash, 2024-12-06:
(Altho if you want to prevent/fix this lower in the stack e.g. in uci, would you prefer to keep this open as another of the 3000 reminders, @jow- ? :)

closed 2024-12-06 by systemcrash