openwrt/luci #7462: luci-app-ttyd: Terminal interface does not work.

zdz2019, 2024-12-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

luci-app-ttyd: Terminal interface does not work.

### Expected behaviour

Under normal circumstances, the terminal interface can perform input and output operations.

### Steps to reproduce

Use the main source code to install luci-app-ttyd,open the terminal interface.

### Additional Information

```shell
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r0-808f027"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r0-808f027"
```

### What browsers do you see the problem on?

Chrome

systemcrash, 2024-12-09:
Works fine for me. You might try deleting your settings and creating new default ones. Is this build from February 2024?

[image]

castillofrancodamian, 2024-12-09:
I got a blank page if I used MbedTLS or OpenSSL. With OpenSSL it works. With the other one I don't remember.

zdz2019, 2024-12-10:
My mistake, wrong configuration.