openwrt/luci #7464: luci: luci in 24-rc2 always config back VHT20 to HT20

lessload, 2024-12-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

when make an change to GL-A1300 wifi
luci always set config back to
`uci set wireless.radio0.htmode='HT20'` 

### Expected behaviour

it should set to `uci set wireless.radio0.htmode='VHT20'`

### Steps to reproduce

1. edit wireless config
2. save
3. luci auto set 2g to HT20

### Additional Information

```shell
NAME="OpenWrt"
VERSION="24.10.0-rc2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0-rc2"
VERSION_ID="24.10.0-rc2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28161-ea17e958b9"
OPENWRT_BOARD="ipq40xx/generic"
OPENWRT_ARCH="arm_cortex-a7_neon-vfpv4"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0-rc2 r28161-ea17e958b9"
```

### What browsers do you see the problem on?

Firefox

Ramon00, 2024-12-11:
Looks like 802.11ac is only defined here  for 5GHz [only:](https://github.com/openwrt/luci/blob/master/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js)
Not sure why that is actually. 802.11ac on 2.4GHz has better performance than 802.11n (e.g. 256-QAM).

@systemcrash any idea?

systemcrash, 2024-12-11:
I saw something about 11.a on the mailing list yesterday and today which could be related. 

http://lists.openwrt.org/pipermail/openwrt-devel/2024-December/043510.html

systemcrash, 2024-12-12:
In any case, I don't think this has anything to do with luci. If it did, you'd see it in the queued changes before applying. @lessload has not provided any such information, so I suspect this is in a lower layer (uci or wireless system).

Ramon00, 2024-12-12:
come to think of it, I guess VHT20 is not allowed on 2.4 GHz...

dannil, 2024-12-12:
VHT is only defined in IEEE 802.11ac-2013 which only encompasses the 5 GHz band, the 2,4 GHz specification is inherited from IEEE_802.11n-2009 which specified it's channels as HT. The question is therefore moot since it isn't a valid configuration anyway.

raenye, 2024-12-14:
See [#12904](https://github.com/openwrt/openwrt/pull/12904).
(if `ipq40xx` uses `ath10k` as the 2.4 wifi driver, it is supported by this PR)