openwrt/luci #7473: Lucy: old link in Network > Interfaces?

janexner, 2024-12-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]
[image]

### Actual behaviour

When I went to Lucy's Network > Interfaces page, there was a notice on my "vpnovh" interface that I should install protocol extensions. When I clicked that link, I got a 404 error.

### Expected behaviour

I think the link should be https://192.168.28.1/cgi-bin/luci/admin/system/package-manager?query=luci-proto rather than https://192.168.28.1/cgi-bin/luci/admin/system/opkg?query=luci-proto

### Steps to reproduce

go to Network > Interfaces. If necessary, create an interface with a not-yet-installed protocol. Click the "Install protocol extensions..." link that will show up.

### Additional Information

```shell
root@ArcherAX23:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="24.10.0-rc2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0-rc2"
VERSION_ID="24.10.0-rc2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28161-ea17e958b9"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0-rc2 r28161-ea17e958b9"
OPENWRT_BUILD_DATE="1733226068"
```

### What browsers do you see the problem on?

Firefox

systemcrash, 2024-12-15:
Closed by 4bc20a2bfb2feca6c9b2e421b89295a0c88353f2

closed 2024-12-15 by systemcrash

hnyman, 2024-12-15:
@jow- 
This issue (and the fix PR #7474 ) brought into attention that there are still some references to opkg in LuCI, although most have been fixed.

One instance that I am unsure about the needed change, is:
https://github.com/openwrt/luci/blob/master/modules/luci-mod-network/htdocs/luci-static/resources/view/network/interfaces.js#L327
`L.resolveDefault(fs.read('/usr/lib/opkg/info/netifd.control'))`

There is no /usr/lib/opkg/ in apk enabled systems... 
Fixing that will likely also generate a difference between LuCI branches: apk-master and opkg-24.10

systemcrash, 2024-12-15:
Was just looking at that one too. Just gotta find the equivalent on an apk system, and make luci-mod-network detect apk/opkg, I think.

janexner, 2024-12-15:
I did not want to touch that. Way over my head.

systemcrash, 2024-12-15:
Actually, @hnyman we may be able to drop that anyway, since it seems to be related to bridge migration. Requires netifd minimum 2021-05-26. Which we are way past on 24.

systemcrash, 2024-12-15:
Should be resolved by e23f6069d6fb91d75147f908bd15d2ee96e6a5d7