openwrt/luci #7475: Feature request: luci-app-keepalived: add `interface_up_down_delays` option

grifos, 2024-12-16:
### What would you like to see in luci?

The maintainer and `keepalived` expert is Florian @feckert 

Would it be possible to add the `keepalived` `interface_up_down_delays` option to `luci-app-keepalived`?

Please note: I realize that to implement this there may also have to be changes to the `keepalived` package itself, for the UCI code (the actual functionality should already be present in `keepalived`). Please just let me know if I should also open a feature request in packages.

This `keepalived` feature allows for the configuration of VRRP master/backup switchover delays to interfaces up or down events.

It was mainly intended to prevent instability due to flapping links, but another major use case is to prevent downtime and unpredictable behavior when using STP, which is my use case.

In my tests, with STP enabled, a WAN side `keepalived` instance will switchover to primary as soon as a tracked LAN interface comes back up, however traffic will not pass due to STP delays. Also, the other VRRP node does not see the main node's VRRP messages and it remains stuck in master mode, even after STP has enabled the link, and only a `keepalived` restart fixes the problem.

The keepalived.conf syntax is as follows:

```
 interface_up_down_delays {
                ifname down_delay [up_delay]
                ifname2 down_delay [up_delay]
                ...
            }
```
More info can be found here: https://www.keepalived.org/manpage.html

Sadly I'm not a programmer, else I'd have been more than happy to submit a PR.

Thank you for all the work you do on OpenWrt. As someone wrote on Phoronix, you guys are doing God's work ;)

feckert, 2024-12-16:
Since I don't have a setup, can you please check if the patch init script creates a valid configuration if you make the following changes to `/etc/init.d/keepalived`  and `/etc/config/keepalived`.

Changes to `/etc/init.d/keepalived`:
```
diff --git a/net/keepalived/files/keepalived.init b/net/keepalived/files/keepalived.init
index cbbff4941..e9d5a4b5e 100644
--- a/net/keepalived/files/keepalived.init
+++ b/net/keepalived/files/keepalived.init
@@ -287,6 +287,25 @@ static_routes() {
        done
 }

+interface_up_down_delay() {
+       local device
+       local down_delay
+       local up_delay
+       local line
+
+       config_get device "$1" device
+       config_get down_delay "$1" down_delay
+       config_get up_delay "$1" up_delay
+
+       [ -z "$device" ] && return
+       [ -z "$down_delay" ] && return
+
+       line="${device} ${down_delay}"
+       [ -z "$up_delay" ] || line="${line} ${up_delay}"
+
+       printf '%b%s\n' "$INDENT_1" "$line" >> "$KEEPALIVED_CONF"
+}
+
 # Count 'vrrp_instance' with the given name ; called by vrrp_instance_check()
 vrrp_instance_name_count() {
        local name
@@ -609,6 +628,10 @@ process_config() {
        config_foreach_wrapper static_ipaddress
        config_section_close

+       config_section_open "interface_up_down_delay"
+       config_foreach_wrapper interface_up_down_delay
+       config_section_close
+
        config_section_open "static_routes"
        config_foreach_wrapper static_routes
        config_section_close
```

Add additional `interface_up_down_delay` section to  to `/etc/config/keepalived`:
```
config interface_up_down_delay
    option device <device>
    option down_delay <number in seconds>
    option up_delay <number in seconds>
```

grifos, 2024-12-16:
Hi Florian, thanks for the very fast reply!

Yes, it creates a valid `keepalived.conf` configuration, here's an excerpt:

```
	smtp_server 127.0.0.1
	router_id r2
}

static_ipaddress {
}

interface_up_down_delay {
	br-lan 10 30
}

static_routes {
}

vrrp_instance wan {
	authentication {
```

grifos, 2024-12-23:
Correction to the above:

The correct syntax for the function is `interface_up_down_delays` (with the trailing "s"), like on the `keepalived` documentation.

feckert, 2025-01-02:
@grifos I'll update the code when I get back to the office.

grifos, 2025-01-02:
Thank you Florian.

One thing to note: the configured delay times have to be less than 2 x the VRRP advert interval, see the below logs. I think it's worth mentioning that on the Luci field.

`Thu Jan  2 17:16:19 2025 daemon.info Keepalived_vrrp[29416]: wan: interface wan debounce timer(s) not less that 2 * advert_int - resetting

Thu Jan  2 17:16:19 2025 daemon.info Keepalived_vrrp[29416]: lan: interface br-lan.101 debounce timer(s) not less that 2 * advert_int - resetting`

feckert, 2025-02-04:
@grifos I have made a pullrequest for this feature.

feckert, 2025-02-05:
@grifos Sorry I forgot to mention the PR https://github.com/openwrt/packages/pull/25893
Could you give them another try?
After this is merged we can integrate this into LuCI.

feckert, 2025-02-13:
@grifos I do not know how to validate the input. What `advert_int` `resetting` values does I have to check?

This is already done by keepalived, hence the info message.
If then I can imagine that we do this in the LuCI?

grifos, 2025-02-13:
Hi Florian, yes that's already done by `keepalived`, my suggestion was only to put a note on the Luci field to let the users know that the timers have to be less than 2 x the VRRP advert interval, in case they don't notice the log entry.

Ideally the Luci field would prevent an out of range timer by comparing the entry with the advert interval entry, but I don't know if that's possible and it wouldn't be worth it if it's too difficult to implement.

feckert, 2025-02-13:
@grifos I have merged the change in keepalived. Now we can add the feature to the luci-app-keepalived

BKPepe, 2026-09-14:
@feckert In this case, we can close this issue, right? :)

closed 2026-09-14 by BKPepe