openwrt/luci #7476: luci-mod-dashboard: wrapping of long hostnames

GrumpyMeow, 2024-12-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

I have some devices with longer hostnames defined. I see that those hostnames get wrapped to fit the column width of the table.

### Expected behaviour

I would expect the hostnames not to wrap as there is enough space available, although currently reserved for the mac- and ip-address.

### Steps to reproduce

Provide a long hostname for a device

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.5"
VERSION_ID="23.05.5"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24106-10cc5fcd00"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.5 r24106-10cc5fcd00"
```

### What browsers do you see the problem on?

Firefox, Chrome, Safari, Microsoft Edge

### Relevant log output

```shell
When looking at the rendering code and output Html i notice that two tables are used for rendering. This makes it hard to get the table styling to adjust to the actual displayed device information (hostname, mac, ip).

https://github.com/openwrt/luci/blob/e23f6069d6fb91d75147f908bd15d2ee96e6a5d7/modules/luci-mod-dashboard/htdocs/luci-static/resources/view/dashboard/include/20_lan.js#L26C3-L36C87
```

adelton, 2025-01-09:
I'd agree that having one table for the headers and one for content, just to force a `<hr/>` between them is not how tables should be done -- see the current DOM:

[image]

adelton, 2025-01-30:
I believe the situation was significantly improved (if not completely fixed) via https://github.com/openwrt/luci/pull/7550.

GrumpyMeow, 2025-06-22:
I just re-installed the luci-mod-dashboard, but still see the wrapping of the hostnames.

[image]

I do notice improvements in the alignment of the column headers. Which is nice.