openwrt/luci #7482: luci: missing encryption information on some wifi networks

damnms, 2024-12-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

I use a banana pi r4 with openwrt master, i build the images myself. 
When i go to the wireless overview, it says on Encryption: - which should be WPA2, as it is wpa2. 

### Expected behaviour

That it shows the correct encryption

### Steps to reproduce

- Get a banana pi r4
- install openwrt master on it
- configure the wireless with encryption turned on
- check the overview

### Additional Information

```shell
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28347-2417770b13"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28347-2417770b13"
OPENWRT_BUILD_DATE="1734516083"
```

### What browsers do you see the problem on?

Firefox, Chrome

systemcrash, 2024-12-18:
@rmandrad maybe you have some ideas?

rmandrad, 2024-12-18:
@damnms can you pls share your /etc/config/wireless ?

damnms, 2024-12-18:
```
root@OpenWrt:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option path 'soc/11310000.pcie/pci0001:00/0001:00:00.0/0001:01:00.0'
	option channel '1'
	option band '2g'
	option htmode 'HT20'
	option cell_density '0'
	option country 'CH'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'soc/11310000.pcie/pci0001:00/0001:00:00.0/0001:01:00.0+1'
	option channel '36'
	option band '5g'
	option htmode 'VHT80'
	option cell_density '0'
	option country 'CH'

config wifi-device 'radio2'
	option type 'mac80211'
	option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option band '2g'
	option channel 'auto'
	option htmode 'HT20'
	option cell_density '0'
	option country 'CH'

config wifi-device 'radio3'
	option type 'mac80211'
	option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+1'
	option band '5g'
	option channel 'auto'
	option htmode 'HE20'
	option cell_density '0'

config wifi-device 'radio4'
	option type 'mac80211'
	option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0+2'
	option band '6g'
	option channel 'auto'
	option htmode 'EHT80'
	option disabled '1'

config wifi-device 'radio5'
	option type 'mac80211'
	option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option radio '1'
	option band '5g'
	option channel 'auto'
	option htmode 'HE40'
	option cell_density '0'
	option country 'CH'

config wifi-device 'radio6'
	option type 'mac80211'
	option path 'soc/11300000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0'
	option radio '2'
	option band '6g'
	option channel 'auto'
	option htmode 'EHT80'
	option cell_density '0'
	option country 'CH'
	option disabled '1'

config wifi-iface 'wifinet0'
	option device 'radio2'
	option mode 'ap'
	option ssid 'oli'
	option encryption 'psk2'
	option key 'supersecretpassword'
	option network 'lan'

config wifi-iface 'wifinet1'
	option device 'radio5'
	option mode 'ap'
	option ssid 'oli5'
	option encryption 'sae-mixed'
	option key 'supersecretpassword'
	option network 'lan'
	option ieee80211w '0'

config wifi-iface 'wifinet2'
	option device 'radio6'
	option mode 'ap'
	option ssid 'oli7'
	option encryption 'sae'
	option network 'lan'
	option key 'supersecretpassword'
	option ocv '0'
	option disabled '1'

```

I also dont understand why i have so many radios ;)

rmandrad, 2024-12-18:
Don't know the history here ...are you using the be14  card? If so it should be only three.

Can you do a 'iw dev'?

I would suggest to do a 1. backup of your wireless config... 2. delete the wireless config file and then issue 3. 'wifi config ' & 4. then use luci to config & validate you still have the same issue.

damnms, 2024-12-18:
```
phy#0
	Interface phy0-ap0
		ifindex 15
		wdev 0x5
		addr 76:5f:50:6d:fe:0a
		ssid oli
		type AP
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 20.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	26994	0	0	0	0	9857191		27392
		Radios: 0 1 2
	Interface phy0.1-ap0
		ifindex 12
		wdev 0x2
		addr 76:5f:50:6d:fe:1a
		ssid oli5
		type AP
		channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
		txpower 23.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcoltx-bytes	tx-packets
			0	0	25082	0	0	0	25457	9338946		25634
		Radios: 1

```

EPinci, 2024-12-23:
I'm seeing the same, also on BPI r4 with 24.10-rc4.

damnms, 2025-03-14:
I reopened it because i have it now too. I use rm /etc/config/wireless && wifi config, then configure the 2.4ghz with minimalistic settings, then the ax network with minimal settings and then i have this wierd behaviour.

minimal settings = defaults from wifi config, just modified the password and set the encryption to wpa2

rmandrad, 2025-03-14:
this is an iwinfo issue ... the function formatWifiEncryption invokes iwinfo

iwinfo shows encryption unknown

damnms, 2025-03-14:
iwinfo shows this on my wifi5 network:

```
phy0.1-ap0 ESSID: unknown
          Access Point: FE:2B:1F:78:BE:6D
          Mode: Master  Channel: 40 (5.200 GHz)  HT Mode: EHT80
          Center Channel 1: 42 2: unknown
          Tx-Power: 20 dBm  Link Quality: 49/70
          Signal: -61 dBm  Noise: -82 dBm
          Bit Rate: 780.0 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

```

in luci it shows the oli5 ssid but not the encryption, this is not a mixed enc , its wpa3 only

rmandrad, 2025-03-14:
> iwinfo shows this on my wifi5 network:
>
> ```
> phy0.1-ap0 ESSID: unknown
>           Access Point: FE:2B:1F:78:BE:6D
>           Mode: Master  Channel: 40 (5.200 GHz)  HT Mode: EHT80
>           Center [...]

interesting, also is on the phy0.1-ap0 that I see the encryption unknown

fridtjof, 2025-03-14:
This is likely the same issue as https://github.com/openwrt/luci/issues/7584 - I found the root cause there