openwrt/luci #7486: firewall: port forward with ipset

vargalex, 2024-12-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

Hi!

I have defined an ipset with match type src_ip. I added some entries (source IP-s) to it. I defined a port forward for my local server using the previously defined ipset. I click Save & Apply and all works as expected. 
When I add a new IP to the ipset and click Save & Apply, than I can reach the server from the new IP too. But, when I delete some IP-s (even just one) from the ipset and click Save & Apply, the server can be reached from the deleted source IPs until I manually restart the firewall. The main problem with this for me, that I need (automatically by a script) to modify this ipset over JSON RPC.

### Expected behaviour

When I modify my ipset entries and click Save & Apply, than from the deleted source IPs should not reach the server.

### Steps to reproduce

1. Go to Network->Firewall->IP Sets
2. Click Add
3. Set a Name for it, change the Packet Field Match to src_ip: Source IP, add some IPs to the ipset
4. Save it
5. Go to Network->Firewall->Port Forwards
6. Click Add
7. Set a Name for it, set the External port, set the Internal IP address, the Internal port and on the Advanced settings tab set the previously saved ipset
8. Click Save & Apply
9. Now you can reach on the External port your internal device from the defined IP addresses
10. Go to Network->Firewall->IP Sets
11. Click edit at the previously defined IP set
12. Delete some IP from it, and when you want, you can add a new one to it
13. Click Save & Apply
14. You can still reach the internal device on the external port from the deleted IP address

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.5"
VERSION_ID="23.05.5"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24106-10cc5fcd00"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.5 r24106-10cc5fcd00"

But the same happens on 24.10.0-rc2.
```

jow-, 2024-12-19:
Try `echo f > /proc/net/nf_conntrack` to flush conntrack. It's likely lingering conntrack entries allowing those deleted IPs to continue working.

vargalex, 2024-12-19:
> Try `echo f > /proc/net/nf_conntrack` to flush conntrack. It's likely lingering conntrack entries allowing those deleted IPs to continue working.

Hi Jow!

Yes, this is not a problem on the router itself (or as I wrote, restarting the firewall). But what can I do over JSON-RPC API?

jow-, 2024-12-19:
You could issue a restart call using object "rc", method "init" and arguments `{ "name": "firewall", "action": "restart" }`

vargalex, 2024-12-19:
I have now tested to flush the conntrack table with your command. The status not changed after that, I can connect from the deleted source IP.

jow-, 2024-12-19:
Ah, then its not conntrack related but due to the fact that the firewall does not repopulate ipsets on reload, which is intentional. You will need to issue an explicit service restart call

vargalex, 2024-12-19:
As I wrote, when I add a new IP to the ipset, than I can reach the service from the newly added source IP but from the deleted IP too. So, I think the reload somehow partially repopulate the ipset...

jow-, 2024-12-19:
Yes, iirc only new entries from configuration are added but existing ones are not purged since the firewall program cannot determine whether those entries have been purged from config or whether they have been added by external programs.

vargalex, 2024-12-19:
OK. Thanks for the explanation. Can I run nft command over JSON RPC API?

vargalex, 2024-12-19:
According to the [JSON RPC API documentation](https://github.com/openwrt/luci/wiki/JsonRpcHowTo) I can directly run `nft` commands over JSON RPC API with system.call method. With this, I can directly remove an IP address from an ip set and add an another IP address to the set. When my IP set name is `test` and the removable IP address is 1.2.3.4, the I must run the `nft delete element inet fw4 test { 1.2.3.4 }'. And with `nft add...` I can add an another IP to the set. With this the changes happen immediately.

AnoBene, 2025-04-01:
Upgrading from 21. to 24.10, found this same issue. 
Wasted a lot of time thinking there is something wrong with wf4/nftables compared to iptables of 21. release. 
Fiddled wrongly with "ipset" command which is for iptables.

Until this bug is not fixed, the UI does not "apply" ipset removals properly, you can run:
/etc/init.d/firewall restart
via ssh which reloads ipsets from config properly.
Not a good solution, to hard reset a whole service per every ipset change needed.

To reiterate the symptoms I see on ver 24.10
I have upgraded all packages (as of 2025.04.01) and still same bug:

On page: (network - firewall - "IP Sets" tab)
/cgi-bin/luci/admin/network/firewall/ipsets

By adding ipsets and entries to existing sets and clicking APPLY.
Check via ssh: nft list sets
Result: All the new ipsets and added entries show up. (working as expected)

By DELETING ipsets and entries in existing sets and clicking APPLY.
Check via ssh: nft list sets
Result: All the DELETED entrys AND ipsets are still active, not actually removed. (NOT working as expected)

Issue described here has nothing to do with "lingering conntrack entries" - which involves firewall rules and testing them.
Issue here is that UI does not actually APPLY deletions/removals of fw4/nftables type "IP sets" and their entries properly.

DerRockWolf, 2025-09-24:
This is not just happening with ipsets, it also happens with normal traffic rules. Disabling or even removing one doesn't actually cause traffic to be blocked.

In this case flushing conntrack using `conntrack -F` fixes it right away.