openwrt/luci #7491: luci-app-firewall: JS error

grosjo, 2024-12-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

When changing parameters in the firewall screen, the screen becomes iresponsive and the console shows multiple errors

### Expected behaviour

Working as usual

### Steps to reproduce

Just install latest 24.10 (git)

### Additional Information

```shell
NAME="OpenWrt"
VERSION="24.10-SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10-SNAPSHOT"
VERSION_ID="24.10-snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28201-5289460fd0"
OPENWRT_BOARD="ath79/generic"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS="no-all mklibs"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10-SNAPSHOT r28201-5289460fd0"
OPENWRT_BUILD_DATE="1734654000"
```

### What browsers do you see the problem on?

Firefox, Chrome

### Relevant log output

```shell
Kindly refer to screenshot
```

lister-wrt, 2024-12-21:
I just ran into this on a master ramips/mt7621 build. I can't apply any changes on the first/main firewall settings page

zdz2019, 2024-12-21:
Related to ES6 commits.

hnyman, 2024-12-21:
Yeah, I was already wondering if these are related to the ES6 test commits from @jow- that @systemcrash picked up from jow's dormant test branch. Not sure it they were yet meant to be in regular use.

systemcrash, 2024-12-21:
@grosjo can you describe how you produced those tracebacks? Did you just load those pages?

chbgdn, 2024-12-21:
@systemcrash It's produced when trying to save settings in wifi or firewall pages.

systemcrash, 2024-12-21:
Did you perform any particular changes or actions? I don't see it by saving those pages.

chbgdn, 2024-12-21:
@systemcrash In firewall page I changed hardware offload and tryed to save.

systemcrash, 2024-12-21:
Can you test a fix for me @grosjo ?

Change
https://github.com/openwrt/luci/blob/b174d53e6638ed975a5d67db9b1e25389a190515/modules/luci-base/htdocs/luci-static/resources/uci.js#L533

to:
```js
 if (c[conf]?.[sid]?.[opt] != null) 
```

js on your device will be compiled down, so download uci from this link.

systemcrash, 2024-12-21:
Should be resolved by 8a422af6a0e97cdb7b3df1049898731a4b2d153d.

systemcrash, 2024-12-21:
Going to leave this open until next builds are done. Looks like rc3 just baked also, so a few more people might encounter this...

hnyman, 2024-12-21:
cc @hauke 
FYI, 24.10.0-rc3 will have somewhat broken LuCI. There will be trouble in saving some settings changes in LuCI. 
(faulty code got merged two days ago, and then the code too too quickly got also into 24.10)

Aborting rc3 and including the fix for rc4 might be an option to consider.

@systemcrash 
The stable branch should start to be stable as the actual release nears. So, we should refrain from getting all code quickly into 24.10, and let the master got forward with new stuff, and then selectively cherry-pick code into 24.10.  Simple option changes and translation changes are more safe to cherry-pick, but this kind of code style changes can be tricky, as this proves.

systemcrash, 2024-12-21:
@hnyman Agreed. Altho I didn't see any info about a new rc being made. In any case, users can just upgrade their luci-base package from package-manager. Those packages are refreshed fairly frequently.

systemcrash, 2024-12-21:
> There might be other similar bugs in the new ES6 code, so wisest might be to revert those commits from 24.10

I do not exclude the possibility, although growing pains will happen. A revert from 24.10 is a safe strategy. The ES6 changes should show any other problems by now, and the changes are out there, so I think it best to tackle any bugs as they turn up, in master. Anything else turns up and I'll revert from 24.10. ( I intentionally separated out the change-set to cover individual files per commit for eventualities like this. )

zdz2019, 2024-12-21:
channel_analysis.js also affected.

[image]

systemcrash, 2024-12-21:
Curious. That doesn't appear related to the ES6 changes. It looks related to other changes.

It speaks of `i` which is in the code, and depends on the previous code where iterator `i` was present but changed to a `forEach` later on.:
```js
		for (var i=0; i < channels.length; i++) {
```

hnyman, 2024-12-21:
Might it be related to the promise graph deletion that you did in f92e07253f22 ?

systemcrash, 2024-12-21:
Doesn't look like it. The prime suspect is 008c1cf356e93711983fcb0afa37485deac47fe5, which I merged in December 2023. But that never got cherry picked to any builds before now, it seems.

systemcrash, 2024-12-21:
It seems the bug crept in initially in ea4119f25c1b70b8b713c4a0b387a731935b00fe where `[i]` become orphaned in a refactor to `forEach`. 🤷

systemcrash, 2024-12-21:
> channel_analysis.js also affected.
> 
> ![Snipaste_2024-12-21_23-22-56](https://private-user-images.githubusercontent.com/50648881/397936488-244e7bb0-3585-4f7e-abcc-858cf994b3f1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ3OTUwMDUsIm5iZiI6MTczNDc5NDcwNSwicGF0aCI6Ii81MDY0ODg4MS8zOTc5MzY0ODgtMjQ0ZTdiYjAtMzU4NS00ZjdlLWFiY2MtODU4Y2Y5OTRiM2YxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIxVDE1MjUwNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE4NTIzYmZhZGE4YzE4NDM1YWFlYWZkNzhlZWU4NWVmMjk5MWI3YjI1ZDUzNzFjZjNmN2QwZjA4MGE4MjEyMGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.7HyKnOvpwjm-nzeqytc6K4xVm_IuD_KYpK92W-I2cHE)

Should be fixed by b43aea743e78bb478495d2bfca046a0417dbdfe7

grosjo, 2024-12-21:
Shall also be pushed to 24.10 branch I beleive

systemcrash, 2024-12-21:
> Shall also be pushed to 24.10 branch I beleive

Already there :) Just needs to get picked up by the build-bots. Might be a a few days.

uklatt, 2024-12-21:
I have a simmilar problem with 
LuCI openwrt-24.10 branch (24.355.00587~f92e072) /[ OpenWrt 24.10.0-rc2 (r28161-ea17e958b9)](https://openwrt.org/)

I am not able to save changes in wireless settings.

Debug console shows:
```
Uncaught TypeError: c[conf][sid] is undefined
    get http://192.168.178.120/luci-static/resources/uci.js?v=24.355.00587~f92e072:38
    cfgvalue http://192.168.178.120/luci-static/resources/view/network/wireless.js?v=24.355.00587~f92e072:109
    parse http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:138
    parse http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:71
    parse http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:264
    parse http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:71
    parse http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:47
    save http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:47
    handleModalSave http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:211
    super http://192.168.178.120/luci-static/resources/luci.js?v=24.355.00587~f92e072:15
    handleModalSave http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:229
    createHandlerFn http://192.168.178.120/luci-static/resources/ui.js?v=24.355.00587~f92e072:369
    attr http://192.168.178.120/luci-static/resources/luci.js?v=24.355.00587~f92e072:106
    create http://192.168.178.120/luci-static/resources/luci.js?v=24.355.00587~f92e072:113
    E http://192.168.178.120/luci-static/resources/cbi.js?v=24.355.00587~f92e072:126
    renderMoreOptionsModal http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:229
    promise callback*renderMoreOptionsModal/< http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:225
    promise callback*renderMoreOptionsModal http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:223
    createHandlerFn http://192.168.178.120/luci-static/resources/ui.js?v=24.355.00587~f92e072:369
    attr http://192.168.178.120/luci-static/resources/luci.js?v=24.355.00587~f92e072:106
    create http://192.168.178.120/luci-static/resources/luci.js?v=24.355.00587~f92e072:113
    E http://192.168.178.120/luci-static/resources/cbi.js?v=24.355.00587~f92e072:126
    renderRowActions http://192.168.178.120/luci-static/resources/view/network/wireless.js?v=24.355.00587~f92e072:99
    renderContents http://192.168.178.120/luci-static/resources/form.js?v=24.355.00587~f92e072:164
```

systemcrash, 2024-12-21:
Yes, it looks like the same problem.

systemcrash, 2024-12-23:
This should be OK by now. rc4 is building and the luci-base package is refreshed.

closed 2024-12-28 by grosjo