openwrt/luci #7494: luci-theme-bootstrap: `ListValue` is not fully displayed in crowded situations

muink, 2024-12-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

In crowded situations, the x-axis of the `ListValue` widget will be compressed making it completely unreadable.

### Expected behaviour

Even in crowded situations, the x-axis of the `ListValue` widget can be fully displayed.
[image]

### Steps to reproduce

Take `23.05.*` `x86_64` as an example:

1. install [luci-app-fchomo](https://github.com/muink/openwrt-fchomo/tree/master/luci-app-fchomo)
   1. install [mihomo](https://fantastic-packages.github.io/packages/releases/23.05/packages/x86_64/packages/mihomo_1.19.0-1_x86_64.ipk)
   2. install [luci-app-fchomo](https://fantastic-packages.github.io/packages/releases/23.05/packages/aarch64_cortex-a53/luci/luci-app-fchomo_git-24.357.16256-6d19a6d_all.ipk)
2. go to: http://192.168.1.1/cgi-bin/luci/admin/services/fchomo/client -> `DNS policy`
3. add section: `{".name": "test", "Type": "domain", "Domain": "a lottttttttt of text........"}`

[image]

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.4"
VERSION_ID="23.05.4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24012-d8dd03c46f"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.4 r24012-d8dd03c46f"
```

### What browsers do you see the problem on?

Chrome, Safari, Microsoft Edge

hnyman, 2024-12-22:
What is fchomo? Not on official package...

Does the problem materialize with other (normal) apps?

Ramon00, 2024-12-22:
This is probably an issue with the table column spacing in that luci-app-fchomo. Can be fixed with by e.g. adjusting/setting the width property  `E('th', {'class': 'th', 'style': 'width:25%'} ` This is unlikely anything to do with the luci-theme-bootstrap itself. I would suggest asking the maintainers of that app if they can adjust the column width.

muink, 2024-12-22:
In fact, I am the main maintainer of luci-app-fchomo. But `luci` does not give a method to specify the column width of `GridSection`.
And [elements](https://github.com/muink/openwrt-fchomo/blob/6d19a6deaa22102dda8368152cebc225ad0af026/luci-app-fchomo/htdocs/luci-static/resources/view/fchomo/client.js#L1007) also follow standard behavior.
And the issue I mentioned does not occur on `luci-theme-material`
[image]

Ramon00, 2024-12-22:
Then check the CSS deltas. But why not just set the widths yourself in the app instead of relying on the default CSS? e.g. with the width property? Or maybe add a linebreak property?

systemcrash, 2024-12-22:
One can specify the overall column width but they must be applied to the individual elements which appear within a grid section. See 

https://openwrt.github.io/luci/jsapi/LuCI.form.ButtonValue.html#width

muink, 2024-12-22:
@systemcrash I tried `4000`, `50%`, `100%`, but it didn't work
``` patch
--- a/luci-app-fchomo/htdocs/luci-static/resources/view/fchomo/client.js
+++ b/luci-app-fchomo/htdocs/luci-static/resources/view/fchomo/client.js
@@ -1012,6 +1012,7 @@ return view.extend({
 		})
 		so.load = L.bind(hm.loadProxyGroupLabel, so, hm.preset_outbound.direct);
 		so.editable = true;
+		so.width = '4000';
 		/* DNS policy END */
 
 		/* Fallback filter START */
```

muink, 2024-12-22:
But if I simply remove `cbi-section-table` from the table
``` javascript
<table class="table cbi-section-table">
```
[image]
Or remove the spaces in `Proxy group`
[image]
Looks like worked

systemcrash, 2024-12-22:
Perhaps that’s the css element responsible for constraining the viewport.

muink, 2024-12-22:
Specifying the width individually for the `ListValue` widget seems to work
[image]

closed 2024-12-22 by muink

Ramon00, 2024-12-22:
Looks like it is applying  [this](https://github.com/openwrt/luci/blob/master/themes/luci-theme-bootstrap/htdocs/luci-static/bootstrap/cascade.css#L2068) or one of the parents or siblings

material:

```
.cbi-section-table-cell {
	line-height: 1.1;
	align-self: flex-end;
	flex: 1 1 auto;
}
```

vs bootstrap:

```
.table.cbi-section-table .td.cbi-section-table-cell {
	white-space: nowrap;
	text-align: right;
}

.table.cbi-section-table .td.cbi-section-table-cell select,
.table.cbi-section-table .td.cbi-section-table-cell .cbi-select {
	width: inherit;
}
```

muink, 2024-12-23:
https://github.com/muink/openwrt-fchomo/commit/f7f7ee42f1018b3cfe29bdaec8267b8a8e0d464c
I set a minimum width for the `ListValue` widgets and the issue was solved.