openwrt/luci #7497: Hi

BaileyCalder, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Actual behaviour

The UTCapitole block list, when selected from the block list sources activates all sub blocklists and disregards the selections further down on the same page. 

Maybe something wrong with this file?
https://github.com/openwrt/packages/blob/b6176b6687cbddc380b8b3e6ccf2d50112a6d2db/net/adblock/files/adblock.categories

### Expected behaviour

Nothing further to explain

### Steps to reproduce

Nothing further to explain

### Additional Information

```shell
NAME="OpenWrt"
VERSION="24.10.0-rc2"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0-rc2"
VERSION_ID="24.10.0-rc2"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28161-ea17e958b9"
OPENWRT_BOARD="bcm53xx/generic"
OPENWRT_ARCH="arm_cortex-a9"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0-rc2 r28161-ea17e958b9"
OPENWRT_BUILD_DATE="1733226068"
```

dibdot, 2024-12-29:
Not reproducible. Provide at least your adblock config and the debug logs of an adblock run.

dibdot, 2025-01-29:
no feedback - closing for now.

closed 2025-01-29 by dibdot