openwrt/luci #7498: luci-base: TypeError: parts[i].startsWith is not a function

1715173329, 2024-12-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

Got the following error:

```
TypeError
parts[i].startsWith is not a function
```

The error code is at:
https://github.com/openwrt/luci/blob/b43aea743e78bb478495d2bfca046a0417dbdfe7/modules/luci-base/htdocs/luci-static/resources/luci.js#L2759

### Expected behaviour

Render the page normally.

### Steps to reproduce

This happens on package `luci-app-aria2`, `luci-app-alist`, and `luci-app-v2raya`'s log page.

After some research I found this seems to be the cause:
https://github.com/openwrt/luci/blob/779ca1412564844efae93e7f00354bf8afd3ac1e/applications/luci-app-aria2/htdocs/luci-static/resources/view/aria2/log.js#L48

If I change it to:
```
'src': L.resource('icons/loading.gif'), 
```
then the page seems to render normally.

I'm not sure if it's a bug of `luci-base` or the code above itself. Anyway it worked in previous version.

### Additional Information

```shell
OpenWrt revision: openwrt/openwrt@975811921e57
LuCI revision: openwrt/luci@b43aea743e78
```

### What browsers do you see the problem on?

Chrome

Ramon00, 2024-12-23:
appears to be [this](https://github.com/openwrt/luci/commit/7c2cde52dbb14d5e878ca775946db9b3f13dcf1d) commit. I guess it needs to be coerced into a string or a `typeof parts[i].startsWith === 'function'` 
@systemcrash can you have a look?

systemcrash, 2024-12-23:
I think I'll guard against this bug and fix the `path()` function, although consistent usage throughout the luci repo is e.g. `L.url("admin", "system", "leds");` and not ` L.url(["admin", "system", "leds"]);`.

Even tho the jsdoc for that function described `string[]`. Weird. I'll fix both....

systemcrash, 2024-12-23:
Closed by 0826411de7c5a107138cf00d1171fde13a1637ec and 7cb2f6584b96d8bbcfa4846a88731d9bc521bd60

Thanks @1715173329 and @Ramon00

closed 2024-12-23 by systemcrash