openwrt/luci #7500: luci:  channel analysis shows nine radios when it should be three

rmandrad, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

1. bananapi4 mediatek/filogic with mt7996 network card
2. three radios shows nine radios when it should be three

### Expected behaviour

1. it should show three interfaces
2. phy0.0-ap0 , phy0.1-ap0, phy0.2-ap0

### Steps to reproduce

1. go to channel analysis

### Additional Information

```shell
NAME="mda-bananapi4"
VERSION="6.6.67"
ID="mda-bananapi4"
ID_LIKE="lede openwrt"
PRETTY_NAME="mda-bananapi4 6.6.67"
VERSION_ID="6.6.67"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28451-53f2300403"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="mda-bananapi4 6.6.67 0"
OPENWRT_BUILD_DATE="1734924847"
```

### What browsers do you see the problem on?

Firefox, Chrome

### Relevant log output

```shell
using developer tools it shows a json rpc call using iwinfo calling for radio0 when it should be the interface phy0.0-ap0

[{jsonrpc: "2.0", id: 19, method: "call",…}]
0
: 
{jsonrpc: "2.0", id: 19, method: "call",…}
id
: 
19
jsonrpc
: 
"2.0"
method
: 
"call"
params
: 
["c8679c13031e4f73d4e9742d4bd3be03", "iwinfo", "info", {device: "radio0"}]
0
: 
"c8679c13031e4f73d4e9742d4bd3be03"
1
: 
"iwinfo"
2
: 
"info"
3
: 
{device: "radio0"}
device
: 
"radio0"
```

Ramon00, 2024-12-24:
> [x] I have searched the existing issues

Not very well apparantly... [7483](https://github.com/openwrt/luci/issues/7483)

systemcrash, 2024-12-24:
Closing in favour of 7483

closed 2024-12-24 by systemcrash