openwrt/luci #7502: LuCI: Realtime graphs buggy after changing browser windows size

ob45, 2024-12-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

Hello and good day!

I've installed the newest OpenWrt Release Candidate RC-4 and have discovered a visual bug by accident.

When opening up any realtime graph on a small browser window and then expanding said window, the displayed info of graph will end up being stuck relative to the bast window size on the "coordinate system". This basically means it ends up staying in the middle and "spawning" new data from there instead of the most right side of the graph.

This ONLY works if you open up any graph in a very narrow browser window and then expand that window. It does NOT work when you open up the graph in a fully sized window and then make it small. In that case you have to reload the winow or switch to a different graph and reload!

This is the graph after opening it up small and then expanding the browser window:
[image]

This is the graph after doing the step above and waiting for a while:

[image]

This is how it should actually look like (always start on the most right side of the "coordinate system"):

[image]

### Actual behaviour

As described on the "screenshots or captures" section

### Expected behaviour

The graph should always start at the most right part of the coordinate system regardless of window size.

### Steps to reproduce

Open up chrome in a very narrow window. Got to Status > Realtime Graphs and wait a couple seconds to fill up the graph. 
After that expand the browser window and make it wide.

### Additional Information

```shell
NAME="OpenWrt"
VERSION="24.10.0-rc4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0-rc4"
VERSION_ID="24.10.0-rc4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28211-d55754ce0d"
OPENWRT_BOARD="ramips/mt76x8"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0-rc4 r28211-d55754ce0d"
OPENWRT_BUILD_DATE="1734915335"
```

### What browsers do you see the problem on?

Chrome

systemcrash, 2024-12-27:
The graph window, and most other windows have a fixed size, which I think is based in the CSS. So this is more of a universal thing. Resizing the window will not affect the graph size (but viewing from a mobile device might).

ob45, 2025-01-02:
I too think that this is a CSS bug. Regarding the "universal thing" it depends what you mean.
Haven't tried this on the phone but considering you can't change (enlarge) the winow size there I don't think It would be an issue on the phone

adelton, 2025-01-09:
@systemcrash It looks like the width of the graph is "fixated" when the `div` / `svg` are initially created. When I resize the browser window and Ctrl+R, the graph correctly resizes to show more (or fewer) minutes of the history.

And I correct assuming that the refresh of the graph is triggered by https://github.com/openwrt/luci/blob/a88d4af4e92bfc7374524c47bd0e53a7d4685a32/modules/luci-base/htdocs/luci-static/resources/luci.js#L758? Have you got a pointer to a place which would describe how these `POST`s fit into the grand scheme of things? And where and how a window size change might best fit in?