openwrt/luci #7504: luci-app-acme: variables don't save using dns-api

IamRPDev, 2024-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

https://github.com/user-attachments/assets/c8b71a71-5198-4520-a31d-7f45f6ea7739

https://github.com/user-attachments/assets/4f7e78a3-e25f-4844-9ef4-ee9ec60b8aaa

### Actual behaviour

I noticed this while setting up cloudflare api, but i tested a few dns api providers and data input also failed to save.

While using the dns-api function, the various new input field don't save the variables.
First Video Example.

But manually adding the variable in the DNS API credentials area such as CF_Token="Example" saves and also displays in the corresponding box.
Second Video Example

I originally experienced this on my MT6000, but I performed a fresh install of the latest snapshot last night on my Netgear R8000 and confirmed this behavior.

### Expected behaviour

I would expect the fields to save the input data, and not just display the manually entered variables, thus making it redundant. 

### Steps to reproduce

1. Go to Services > ACME Certificates
2. Add cert config
3. Change Validation method to DNS
4. Go to DNS Challenge Validation tab
5. Change DNS API to CloudFlare.com
6. Enter Data. Into any of the Fields. 
   i.e.CF Key, CF Email, CF Token, CF Account ID, CF Zone ID
7. Save
8. Edit new entry to view empty fields.

### Additional Information

```shell
AME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28576-a9895c4585"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS="busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28576-a9895c4585"
OPENWRT_BUILD_DATE="1734722468"
```

### What browsers do you see the problem on?

Firefox

systemcrash, 2024-12-26:
Are you using something like lastpass or 1password web browser plugin?

IamRPDev, 2024-12-26:
@systemcrash I don't run either add-on, but to be sure I started a new profile on firefox without any addons. I also reflashed my R8000 with a fresh download of the snapshot. Installed only luci and luci-app-acme and acme-acmesh-dnsapi.

**I retested and still experienced the same issue. Updated info below.** 

I am also including my OS info and Browser version because I am running linux. I can't see how that would matter, but who knows if the linux version for gecko/firefox is the reason. For that matter I made sure to run apt update before testing as well.

https://github.com/user-attachments/assets/62b583cf-abab-47ff-91d0-e2a27dfdf049

**Updated cat /etc/os-release for second test**

NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28406-d1e9c50d06"
OPENWRT_BOARD="bcm53xx/generic"
OPENWRT_ARCH="arm_cortex-a9"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r28406-d1e9c50d06"
OPENWRT_BUILD_DATE="1735035828"

**OS INFO**

NAME="Linux Mint"
VERSION="21.3 (Virginia)"
ID=linuxmint
ID_LIKE="ubuntu debian"
PRETTY_NAME="Linux Mint 21.3"
VERSION_ID="21.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=virginia
UBUNTU_CODENAME=jammy

**Browser Versions**

firefox --version
Mozilla Firefox 133.0.3

librewolf --version
Mozilla Firefox 133.0.3-1
**Including because its the original browser I detected this on. but should still be the same version of gecko**

systemcrash, 2024-12-26:
Thanks for checking. This is definitely a bug 🐛 

I know what’s going wrong and might have a fix. 

In the meantime, @stokito.

systemcrash, 2024-12-26:
So with how @stokito did everything in 32c3fd4922db7c16fc2ef4040ae49ba6a7db57cb, possibly without realising, currently the values entered will never be saved themselves, as they're only stored in the 'credentials' setting. ( Since the init scripts read that variable ). But the page sequentially writes each of the CF_ variables, so the 'last write wins'. Only ever one is stored. 

To handle that, it needs to load the existing values during save to be able to add to them. But once it loads existing values, it becomes impossible to delete existing values due to a data race condition. So the page needs a rewrite to avoid all of this.

The only reliable way to store those values and the processed 'credentials' is during the write of 'credentials' variable itself which needs to handle/check every provider variable during write time...

IamRPDev, 2024-12-27:
@stokito Unfortunately I'm experiencing the same issue.

systemcrash, 2024-12-27:
If you remove the credentials field, then you cannot delete them or modify them later (not easily), which defeats the point of the GUI. Welcome to the data race condition :)

IamRPDev, 2024-12-27:
@stokito Just like @systemcrash said if you remove that line you can enter data, but not remote it.

IamRPDev, 2024-12-27:
@systemcrash @stokito Also it seems if you create the cert config before changing the account email on the initial page it does not update. Also if you delete the config and create it again it keeps the previous email. You need to delete the folder in /etc/acme/*domainname* manually to get the values to update. This might also apply to other values, i didn't get a chance to test it fully.

IamRPDev, 2024-12-27:
If there is anything you'd like me to test let me know. While I'm still learning the programming end of things, I'm a capable tech. 

I'd also love to support the project anyway I can, giving back where possible.

IamRPDev, 2024-12-29:
@stokito Looking good, edits come through. Only exemption being that a blank entry does not delete the previous variable when editing the cert config. If I delete the accidental entry in CF Account ID it doesn't remove the CF_Account_ID variable. See Example 1 below. Although if I delete the variable it does delete the CF Account ID.
Example 2 below

**What I did find broken.** I noticed is that changing the email on the frontpage (option account_email 'letsencrypt@example.org') does not change the variable (CA_EMAIL) in 
/etc/acme/ca/acme-v02.api.letsencrypt.org/directory/ca.conf or in account.json. 

I noticed this because the email I used at first was cert2@mydomain.xyz. Then I decided to break it and entered test@example.org, noticed it didn't break. Then I went digging in the configs and noticed the original email. I changed it to the email I actually use which is letsencrypt@mydomain.xyz and still no change.

**Definitely something to look into. Example 3 Below** 
(I edited the data to preserve security)

**Example 1**
[image]

**Example 2**
[image]

**Example 3**
[image]

systemcrash, 2024-12-29:
I've just tested the code and it works satisfactorily, so I'll wait for your word @IamRPDev.

IamRPDev, 2024-12-29:
@stokito @systemcrash 

The Cert generation looks perfect now, the realtime update is impressive.
I would say as far as the original ticket goes its corrected the issue.

**I'm more than happy to open another ticket for the account ID issue, but if you'd like to ignore this for now I'd understand but please see my argument for it below.**

As for the Account ID. It does not regenerate. I can delete the original cert config and folder in /etc/acme that i was trying to generate, and try generating a key for a whole new domain. It still doesn't regenerate a new user key.

The biggest issue I have with this is naturally I'd change email and debug , then i'd go to setup a cert config. When saving the cert config it resets the changes made to the email and debug field. But by then I've scrolled down allowing me to miss them changing back. So it easily allows you to save the incorrect email when you hitting Save & Apply. Thus locking in the changes until I delete the files in /etc/acme/ca/acme-v02.api.letsencrypt.org/directory/. I'm not confident someone using the gui to know where this is, and it permanently breaks the key generation. Plus if someone wanted to change the email they were using it changes in the gui but not on the backend.

My only recommendation on this would to delete the user CA config when the email is change that way it regenerates upon the acme service being restarted. Deleting the files in /etc/acme/ca/acme-v02.api.letsencrypt.org/directory/ seems to accomplish this.

I'll let you both make this decision.

IamRPDev, 2024-12-29:
@stokito I can see your point. I figured when changing the email it would deploy a new cert not try to renew. I hadn't made it that far into testing today when I hit my max cert creation for the next few days. 

I'll close this ticket for the initial issue. Thanks for all the work you put in!

PR #7510 (2024-12-30): luci-app-acme: DNS API store all params
Fix  #7504
commit f843014f (2024-12-30): luci-app-acme: DNS API store all params

Previously only one DNS API field was stored.
Get the current creds list, alter it and save.
Fix #7504
closed 2024-12-29 by IamRPDev