openwrt/luci #7506: LuCi: DHCP and DNS option fails

rastersoft, 2024-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

I went to DHCP and DNS option in OpenWrt 24.10.0-rc4 and it shows that error. It works fine in OpenWrt 23.05.5

### Expected behaviour

It should show the DHCP and DNS screen.

### Steps to reproduce

Go to Network -> DHCP and DNS

### Additional Information

```shell
NAME="OpenWrt"
VERSION="24.10.0-rc4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0-rc4"
VERSION_ID="24.10.0-rc4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28211-d55754ce0d"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0-rc4 r28211-d55754ce0d"
OPENWRT_BUILD_DATE="1734915335"
```

### What browsers do you see the problem on?

Firefox

systemcrash, 2024-12-25:
Please paste the output from the browser console. (Ctrl+alt+k or so?)

rastersoft, 2024-12-25:
Here it is:

```
TypeError: nodes.querySelector(...) is null
    renderContents https://192.168.0.251/luci-static/resources/view/network/dhcp.js?v=24.357.59006~41cbd31:49
    promise callback*render/s.renderContents https://192.168.0.251/luci-static/resources/view/network/dhcp.js?v=24.357.59006~41cbd31:49
    promise callback*render https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:159
    renderChildren https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:34
    renderContents https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:48
    promise callback*render https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:48
    render https://192.168.0.251/luci-static/resources/view/network/dhcp.js?v=24.357.59006~41cbd31:112

[luci.js:154:9](https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31)
Uncaught (in promise) TypeError: nodes.querySelector(...) is null
    renderContents https://192.168.0.251/luci-static/resources/view/network/dhcp.js?v=24.357.59006~41cbd31:49
    promise callback*render/s.renderContents https://192.168.0.251/luci-static/resources/view/network/dhcp.js?v=24.357.59006~41cbd31:49
    promise callback*render https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:159
    renderChildren https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:34
    renderContents https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:48
    promise callback*render https://192.168.0.251/luci-static/resources/form.js?v=24.357.59006~41cbd31:48
    render https://192.168.0.251/luci-static/resources/view/network/dhcp.js?v=24.357.59006~41cbd31:112
[luci.js:49:647](https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31%20line%20171%20%3E%20eval)
    error https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31:158
    (Asíncrono: promise callback)
    __init__ https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31:139
    super https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31:15
    ClassConstructor https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31:5
    compileClass https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31:175
    (Asíncrono: promise callback)
    compileClass https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31:171
    (Asíncrono: promise callback)
    require https://192.168.0.251/luci-static/resources/luci.js?v=24.357.59006~41cbd31:177
    instantiateView https://192.168.0.251/luci-static/resources/ui.js?v=24.357.59006~41cbd31:369
    <anonymous> https://192.168.0.251/cgi-bin/luci/admin/network/dhcp:50
    (Asíncrono: promise callback)
    <anonymous> https://192.168.0.251/cgi-bin/luci/admin/network/dhcp:49
```

rastersoft, 2024-12-25:
And as capture:
[image]

systemcrash, 2024-12-26:
Please share the contents of your `/etc/config/dhcp` file. You can redact any settings you think sensitive.

systemcrash, 2024-12-26:
At a guess it appears as though you might not have any dnsmasq section entries (or they were deleted?).

systemcrash, 2024-12-26:
Until there is better handling of this situation, you can add the following to your `/etc/config/dhcp`
```
config dnsmasq
```

And go to the GUI and tweak your settings.

rastersoft, 2024-12-26:
Confirmed: adding that "fixes" the problem. Thanks!

Andreotti100, 2025-02-07:
Same problem here on my Netgear r6260.

I don't need DHCP, so I disabled it by the checking "Ignore interface" at Interfaces->Lan->DHCP Server,
Upgraded from 23.05.5 to 24.10.0
Then the error appeared when accessing "DHCP and DNS" settings.

Adding "config dnsmasq" to the end of "/etc/config/dhcp" fixed it.

YodaDaCoda, 2025-02-09:
I'd removed dnsmasq and replaced it with odhcpd+unbound. Removing dnsmasq hid the `DHCP and DNS` menu entry under the `Network` item, preventing me from managing my static leases.

Re-installing dnsmasq and rebooting returned the menu entry, but then I encountered this issue. 

Adding `config dnsmasq` to the end of `/etc/config/dhcp` fixed it for me.

prajaybasu, 2025-11-27:
Seems like https://github.com/openwrt/luci/commit/7e1b545bfa1d98a72d18c39528dade2b35421ed5 should allow access to the DHCP tab without dnsmasq. Till then, on 24.10 I just did `touch dnsmasq` in `/usr/sbin` to make the DHCP and DNS menu item load without installing the dnsmasq package.