openwrt/luci #7507: luci-app-pbr: when restarting pbr, web interface hangs

mikhirev, 2024-12-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

1. Click the "restart" button in the pbr page.
2. The page is shadowed waiting for sefvice restart.
3. Nothing changes until I refresh the page.

### Expected behaviour

The page returns to normal mode after service has restarted.

### Steps to reproduce

1. Go to Services → Policy based routing
2. Click "Restart"

### Additional Information

```shell
NAME="OpenWrt"
VERSION="24.10.0-rc4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0-rc4"
VERSION_ID="24.10.0-rc4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28211-d55754ce0d"
OPENWRT_BOARD="ath79/nand"
OPENWRT_ARCH="mips_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0-rc4 r28211-d55754ce0d"
OPENWRT_BUILD_DATE="1734915335"
```

### What browsers do you see the problem on?

Firefox

systemcrash, 2024-12-26:
Are there any errors output to the browser console?

systemcrash, 2024-12-26:
Is this behaviour consistent, or just a one time occurrence?

Please share your `/etc/config/pbr`. You're config change might isolate you from the router temporarily.

mikhirev, 2024-12-26:
> Are there any errors output to the browser console?

Yep.

```
Uncaught (in promise) Error: XHR request timed out
    handleReadyStateChange https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:62
luci.js:62:20
```

> Is this behaviour consistent, or just a one time occurrence?

It is consistent.

> Please share your `/etc/config/pbr`.

```
config pbr 'config'
        option verbosity '2'
        option src_ipset '0'
        option dest_ipset '0'
        option resolver_set 'dnsmasq.nftset'
        option ipv6_enabled '0'
        list ignored_interface 'vpnserver wgserver'
        option boot_timeout '30'
        option rule_create_option 'add'
        option procd_reload_delay '1'
        option webui_protocol_column '0'
        option webui_chain_column '0'
        option webui_show_ignore_target '0'
        option webui_sorting '1'
        list webui_supported_protocol 'tcp'
        list webui_supported_protocol 'udp'
        list webui_supported_protocol 'tcp udp'
        list webui_supported_protocol 'icmp'
        list webui_supported_protocol 'all'
        option strict_enforcement '0'
        list supported_interface 'wireguard'
        option enabled '1'
        option webui_enable_column '1'

config include
        option path '/etc/pbr.netflix.user'
        option enabled '0'

config include
        option path '/etc/pbr.aws.user'
        option enabled '0'

config policy
        option name 'google play'
        option dest_addr 'play.google.com'
        option interface 'wireguard'

config policy
        option name 'wiki loves earth'
        option dest_addr 'wikilovesearth.org'
        option interface 'wireguard'

config policy
        option name 'analog devices'
        option interface 'wireguard'
        option dest_addr 'analog.com'

config policy
        option name 'intel'
        option dest_addr 'intel.com'
        option interface 'wireguard'
```

I omitted several "policy" entries. BTW I've just upgraded from OpenWRT 21.02 where I used `vpn-policy-routing`, there are almost no config changes since that.

systemcrash, 2024-12-26:
Roughly how long does the operation take? Try to restart pbr from the command line. Something like `service pbr restart`.

mikhirev, 2024-12-26:
It takes 32 seconds.

systemcrash, 2024-12-26:
I think 30 seconds is the default time out…

mikhirev, 2024-12-26:
A bit more verbose error from browser console:

```
Uncaught (in promise) Error: XHR request timed out
    handleReadyStateChange https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:62
    request https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:40
    request https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:40
    promise callback*request https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:40
    flushRequestQueue https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:32
    request https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:43
    request https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:40
    promise callback*request https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:40
    post https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:64
    call https://192.168.1.1/luci-static/resources/rpc.js?v=24.358.65051~7cb2f65:5
    declare https://192.168.1.1/luci-static/resources/rpc.js?v=24.358.65051~7cb2f65:23
    declare https://192.168.1.1/luci-static/resources/rpc.js?v=24.358.65051~7cb2f65:20
    setInitAction https://192.168.1.1/luci-static/resources/pbr/status.js?v=24.358.65051~7cb2f65:5
    click https://192.168.1.1/luci-static/resources/pbr/status.js?v=24.358.65051~7cb2f65:13
    attr https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:106
    create https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:113
    E https://192.168.1.1/luci-static/resources/cbi.js?v=24.358.65051~7cb2f65:126
    render https://192.168.1.1/luci-static/resources/pbr/status.js?v=24.358.65051~7cb2f65:13
    promise callback*render https://192.168.1.1/luci-static/resources/pbr/status.js?v=24.358.65051~7cb2f65:5
    render https://192.168.1.1/luci-static/resources/view/pbr/overview.js?v=24.358.65051~7cb2f65:9
    promise callback*__init__ https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:139
    super https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:15
    ClassConstructor https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:5
    compileClass https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:175
    promise callback*compileClass https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:171
    promise callback*require https://192.168.1.1/luci-static/resources/luci.js?v=24.358.65051~7cb2f65:177
    instantiateView https://192.168.1.1/luci-static/resources/ui.js?v=24.358.65051~7cb2f65:369
    <anonymous> https://192.168.1.1/cgi-bin/luci/admin/services/pbr:50
    promise callback* https://192.168.1.1/cgi-bin/luci/admin/services/pbr:49
luci.js:62:20
```

systemcrash, 2024-12-26:
Peek around in the console when you've pressed apply. Look at the headers of the XHR POST request.

stangri, 2025-03-06:
@mikhirev thanks for reporting this, having said that, 32 seconds to start pbr is unusual.

@stokito I'd welcome any help handling longer than 30 seconds RPCD calls both for luci-app-pbr and luci-app-adblock-fast.

stangri, 2025-04-08:
@mikhirev just to double-check, you're using the nft flavour of pbr, right?

mikhirev, 2025-04-09:
I don't know what flavours pbr has. I use a package from openwrt repo.

mikhirev, 2025-04-09:
pbr	1.1.8-r4

stangri, 2025-04-10:
@systemcrash thanks!
@mikhirev that would be the nft-only version, the reason I asked, there was an issue in luci app which was fixed when running the iptables-variant of pbr, wanted to check if your issue is unrelated.

Menion2k, 2025-05-17:
Similar issue here

OpenWrt 24.10.1, once installed luci-app-pbr the "Overview" page loads very slow and endup in "?" at most of the system information and the rest of the page is broken

stangri, 2025-05-18:
> Similar issue here
> 
> OpenWrt 24.10.1, once installed luci-app-pbr the "Overview" page loads very slow and endup in "?" at most of the system information and the rest of the page is broken

what are the pbr and luci-app-pbr versions?

does the pbr page itself load fine?

if you want to fix the overview page, you can delete the `/www/luci-static/resources/view/status/include/72_pbr.js` file. You might have to also run `rm -rf /var/luci-modulecache/; rm -f /var/luci-indexcache;` to clear luci cache and delete your browser cache for OpenWrt device IP.