openwrt/luci #7519: luci-base: Sorting of `GridSection` does not work

muink, 2025-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]

### Actual behaviour

Click on any table title will not trigger sorting.

### Expected behaviour

Click on any table title to trigger sorting

### Steps to reproduce

Click on any table title.

### Additional Information

```shell
NAME="OpenWrt"
VERSION="24.10.0-rc4"
ID="OpenWrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 24.10.0-rc4"
VERSION_ID="24.10.0-rc4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r28211-d55754ce0d"
OPENWRT_BOARD="x86/64"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 24.10.0-rc4 r28211-d55754ce0d"
OPENWRT_BUILD_DATE="1734915335"
```

### What browsers do you see the problem on?

Firefox, Chrome, Safari, Microsoft Edge

### Relevant log output

```shell
form.js?v=24.365.05279~f843014:215 Uncaught TypeError: Assignment to constant variable.
    at ClassConstructor.eval (form.js?v=24.365.05279~f843014:215:314)
    at NodeList.forEach (<anonymous>)
    at ClassConstructor.handleSort (form.js?v=24.365.05279~f843014:215:85)
    at ClassConstructor.eval (ui.js?v=24.365.05279~f843014:369:29)
eval	@	form.js?v=24.365.05279~f843014:215
handleSort	@	form.js?v=24.365.05279~f843014:215
eval	@	ui.js?v=24.365.05279~f843014:369
```

dannil, 2025-01-01:
See #7520

systemcrash, 2025-01-01:
Closed by f5c18061ed3cd5333ff0b034793d9ae08838ceb6. The build-bots will pick this up in the next few days.

closed 2025-01-01 by systemcrash