openwrt/luci #7522: luci-app-firewall OR luci-app-network: Can create firewall zone in interfaces of any length but firewall allows only 11 chars

Philip2809, 2025-01-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### screenshots or captures

[image]
[image]

### Actual behaviour

Create a firewall zone via the interface in /cgi-bin/luci/admin/network/network then go to /cgi-bin/luci/admin/network/firewall and edit this newly created zone, you will get an error to set the name to something less than 12 chars.

### Expected behaviour

Either to allow a max of 11 chars in the -- custom -- field in network settings or remove max 11 char limit in firewall. There does not seam to be a limit on the number of characters. Also if I do some other error, like invalid characters then I get no error message or feedback, it just does not save. 

[image]

### Steps to reproduce

1. Go to /cgi-bin/luci/admin/network/network
2. Create interface and create firewall from the interface in the "Firewall settings" tab
3. Name it more then 11 chars
4. Save
5. Go to /cgi-bin/luci/admin/network/firewall
6. Edit the newly created firewall zone and see the error message on the name field

### Additional Information

```shell
NAME="OpenWrt"
VERSION="23.05.4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.4"
VERSION_ID="23.05.4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r24012-d8dd03c46f"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.4 r24012-d8dd03c46f"
```

### What browsers do you see the problem on?

Firefox, Chrome, Safari, Microsoft Edge

hnyman, 2025-01-02:
The size limitation comes from kernel & netfilter rules, so the limitation is quite real and needed.

The same validation should likely then be assigned to the create dialog, too.

adelton, 2025-01-09:
I gave the additional validation a try in https://github.com/openwrt/luci/pull/7549. Hopefully the widget is the right place to do it without negative side-effects.