openwrt/luci #7528: „Weblate“, a lot of errors, missing files, old files conflicting, as they no longer exist. (Migration problem to 24.X)

dziugas1959, 2025-01-05:
### Is there an existing issue for this?

- [X] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]
[image]
[image]

### Actual behaviour

Errors are shown, appeared a month ago.

### Expected behaviour

None of these should appear.

### Steps to reproduce

1. go to: https://hosted.weblate.org/projects/openwrt/

### Additional Information

```shell
„Weblate 5.10-dev“
```

### What browsers do you see the problem on?

Firefox

systemcrash, 2025-01-05:
This is the result of removing the English translations, since they're identical to the GUI.

dannil, 2025-01-06:
Do we need a maintainer with Weblate access to configure it properly then? I assume it's a consequence of https://github.com/openwrt/luci/commit/0ca52ebe27e96f01b82250e58a375bac0def9561?

hnyman, 2025-01-06:
It is due to that commit, but it just highlighted the underlying problem that a dozen apps in Weblate have been configured to have en .po file as template, instead of proper templates .pot

It requires manual config for those faulty apps in Weblate config.

hnyman, 2025-01-06:
But so far 24.10 is already equal to master, so there is no backport need.
(They will likely deviate soon, but right now they are equal)

hnyman, 2025-01-09:
I fixed the wrong template path

dziugas1959, 2025-01-09:
> I fixed the wrong template path

Still a phantom file exists, that every language shows an error for:
[image]
[image]
Only a maintainer can fix this, this was addressed in this post, though not fixed, even when marked here as „completed“.

hnyman, 2025-01-09:
Please note that some of the warnings are shown for historical reasons, e.g. in your screenshot "appeared month ago, last seen a month ago".
So some error conditions / displays also clear themselves via time.

But I still re-opened the issue as you wish.

dziugas1959, 2025-01-09:
From my understanding, this is a bug in „Weblate“, it thinks that there is a file, that no longer exists, as such, shows an error.

hnyman, 2025-01-09:
> „Weblate“, it thinks that there is a file, that no longer exists, as such, shows an error

It thought so, a month ago. **But not any more.**  Like it says in the screenshots.
To my understanding, it just shows the error message for a while for some reason. Strange, but still.

[image]

I do not find any actual errors in the system admin console. 
Only the Glossary component shows the error signs, but when going there, all error notices leads to this same Lithuanian thing...

And naturally we have no real insight into Weblate internals. We are just a downstream user of the service.

dziugas1959, 2025-01-11:
> It thought so, a month ago. **But not any more.** Like it says in the screenshots. To my understanding, it just shows the error message for a while for some reason. Strange, but still. [...]

Try doing what the error message says on top of that: „_Please fix this by removing duplicated strings with same identifier from the translation files._“
Most likely, this was a string, that was duplicated with the English (non-developer) and I just copied the glossary translation, which was fine, until English (non-developer) was removed, causing this double phantom string to exist.

Try checking the ID for the place of this string (_IPv6_) and removing it, then re-adding it. My guess, why it doesn't show an error in administration, is most likely because, it doesn't exist and „Weblate“ developers did not account for this type of bug to happen.

dziugas1959, 2025-01-18:
New problems, this time with the Firmware Wizard, Bulgarian.

[image]

hnyman, 2025-01-18:
Actually, like I predicted, the announcements about old errors got timed out and went away.
The old config related error (that this issue report is about) was fixed by me last week.

Now there is just a "normal" conflict.
I have no access to Firmware wizard repo, so I sent a note to @aparcar. Also @jow- should be able to fix that.

closed 2025-01-18 by hnyman

systemcrash, 2025-01-18:
Actually, I opened a ticket with weblate 😉

They opened this: 

https://github.com/WeblateOrg/weblate/issues/13511

dziugas1959, 2025-01-18:
Ideally, issues should only be closed when fixed, since the entire component is locked from any edits. Hopefully this will be fixed in a timely manner.

hnyman, 2025-01-18:
> the entire component is locked from any edits. Hopefully this will be fixed in a timely manner.

Feel free to author a fix PR for the firmware wizard git repo, and then hope that somebody with merge rights there merges it in. You have just as much power in that repo as I do.