openwrt/luci #7535: Feature request: Labels/comments for MAC Addresses (e.g. for WiFi MAC Address filters)

aragilar, 2025-01-08:
### What would you like to see in luci?

Currently there are a number of fields where one might input a MAC Address, but where the meaning of the MAC Address lacks some context (e.g. in the list of MAC Addresses in WiFi MAC filtering tab). Would it be possible to introduce a comment/description/label field in such places?

adelton, 2025-01-11:
What type of context do you envision being added? Specifically in the MAC-Filter tab there is `Allow listed only` and `Allow all except listed` (besides the `disable`) which pretty much defines the context for the MAC-List.

aragilar, 2025-03-04:
It's definitely similar. I'm not sure what the motivation in #7551 is, but mine is basically the equivalent of sticking comments in the underlying text files, and I don't expect them to appear outside where the MAC Addresses are entered (i.e. not a hostname alternative).

closed 2025-03-04 by hnyman

aragilar, 2025-03-10:
Ah, I think you've confused what was in this issue vs #7551. I'm not trying to give a independent name to MAC addresses, I'm looking at adding effectively an accompanying free text field so comments can be seen in the web UI. Here's an example of how I think the config file would look (as far as I understand there are no lists of lists in uci?):

```
config wifi-iface 'default_radio0'
        option macfilter 'allow'
        list maclist '00:00:00:00:00:00'
        list maclist-comment 'We added this MAC for Bob to connect blah'
        list maclist '00:00:00:00:00:01'
        list maclist-comment 'We added this MAC so Alice can monitor wifi in the second building'
```

Does this make more sense (I figure there's lots of other places where having comments like this would be useful, but I was trying to suggest a smaller initial feature)?