openwrt/luci #7536: luci-app-keepalived: scripts broken

slaclau, 2025-01-08:
### Is there an existing issue for this?

- [X] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

After adding a vrrp script, the scripts page hangs and displays the error above

### Expected behaviour

The page should allow a track script to be added

### Steps to reproduce

Add a vrrp script

### Additional Information

```shell
{
        "kernel": "6.6.69",
        "hostname": "router_2",
        "system": "Intel(R) Celeron(R) N5095 @ 2.00GHz",
        "model": "QEMU Standard PC (Q35 + ICH9, 2009)",
        "board_name": "qemu-standard-pc-q35-ich9-2009",
        "rootfs_type": "ext4",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0-rc5",
                "revision": "r28304-6dacba30a7",
                "target": "x86/64",
                "description": "OpenWrt 24.10.0-rc5 r28304-6dacba30a7",
                "builddate": "1736026537"
        }
}
```

### Relevant log output

```shell
i is not defined in renderTrackScript
```

feckert, 2025-01-08:
I can not reproduce this issue.
[image]

dannil, 2025-01-08:
https://github.com/openwrt/luci/blob/ab1bf13c6ea4c10cf7009d7919d9f11eed8bd4b3/applications/luci-app-keepalived/htdocs/luci-static/resources/view/keepalived/script.js#L37-L39

This isn't valid JavaScript in strict mode, the `i` variable needs to be declared with `let`, which is why it's crashing.

feckert, 2025-01-09:
I have merged the PR that fixes the issue.
The only thing I don't understand is why it works for me?

closed 2025-01-09 by feckert

dannil, 2025-01-09:
> I have merged the PR that fixes the issue. The only thing I don't understand is why it works for me?

Unsure, are you running a vanilla 24.10.0-rc4 or newer build or do you have your custom image? Maybe you've configured something so that the `'use strict'` definition as defined in the JavaScript sources are ignored?