openwrt/luci #7539: Feature request: WireGuard interfact should allow for IPv6-PD intput

adelton, 2025-01-08:
### What would you like to see in luci?

When configuring a 6in4 interface with LuCI, it is possible to specify `IPv6 routed prefix` in the `General Settings` tab. The value(s) then lands in `/etc/config/network` in `list ip6prefix`.

For the WireGuard VPN protocol, the same UI-based approach is desired. Configuring the `ip6prefix` manually in `/etc/config/network` works and is then shown in the `IPv6-PD` field on the interface overview page but making it part of the UI would make things easier.

amanjuman, 2025-02-28:
This is very much needed.
I wrote a guide for manual setup [here](https://medium.com/@amanjuman/setting-up-wireguard-ipv6-tunnel-with-prefix-delegation-pd-on-openwrt-a56d7d6c6ff9).

vevere, 2025-05-21:
thanks a lot for the tip (set `ip6prefix `manually in `/etc/config/network`) ... you save my day !
for those who want to easily have their entire LAN behind a VPN that is great