openwrt/luci #7553: WiFi: mandate WPA3 in >= 6GHz bands

systemcrash, 2025-01-10:
I want to get some feedback before adding a fix for this. It's my understanding that 6GHz bands (WiFi 6E and beyond) mandate WPA3 and OWE only. See [here](https://www.extremenetworks.com/resources/blogs/wireless-security-in-a-6-ghz-wi-fi-6e-world) and [here](https://www.intel.com/content/www/us/en/products/docs/wireless/wi-fi-7.html#editorialTableBlade-1). Even there may be some devices which don't care about this. 

[Forum Example](https://forum.openwrt.org/t/adding-support-for-linksys-mr7500/170721/56).

How does the One (with WiFi 7) behave here? @Ansuel @aparcar 

WiFi 7 in general @rmandrad ?

A 'force' option as the new default is possible, so those wanting older WPA2 in 6GHz can have it.

rmandrad, 2025-01-10:
@systemcrash thank you for asking for my feedback. From a user perspective similar to 6e WPA3 (OWE haven't tried it) WIFI7 requires these. The  banapi4 wifi forum has plenty of examples of users falling on this "trap"...

My observation, there are some dependencies - 1. WPA3/OWE if possible, should be forced through the mac80211.sh / hostpad.sh scripts (if not already) or the new ucode implementation. Also, 2. firstboot setup should specify wpa3 (with a default password?)  somehow through the json spec of the device ... probably very hard ...

regarding luci - yes if it is wifi7 other than wpa3/owe the other options should not be available.

worth perhaps also include on this @blogic as he probably has an updated view of the ucode changes and if there is anything on this backlog regarding your question

ps. I haven't heard about a force option to use WPA2 or mixed for wifi7 ... why one would allow this ? as it is against the standard so imho i would keep away of implementing this.

systemcrash, 2025-01-10:
@rmandrad Could you give [this change](https://github.com/systemcrash/luci/blob/36e14ec0ed04bee26cbf8cf677bbdcf7e93188fa/modules/luci-mod-network/htdocs/luci-static/resources/view/network/wireless.js) a spin and verify it works as intended on a 6G band?

rmandrad, 2025-01-11:
so tested

Test approach

replaced wireless.js, removed the luci-indexcache file and restarted the web server.

Test - Using the  banana pi4 device - with a mt7996 card, radio2 is 6ghz ,operating frequency mode BE  only WPA3* and OWE are shown.
[image]

I believe there is an issue on the code - it shouldn't be the band

`const twoOrFiveGBand = (band == "2g" || band == "5g");`

but instead the operating mode BE (EHT) that should force the use of WPA3 or OWE.

I can use AX on the 6G radio, equally I can use EHT on the 5Ghz radio (radio1).

If I select the radio1 and BE I am not constrained anymore to use only WPA3 and OWE

Ramon00, 2025-01-11:
> Yeah, I just don't know the plethora of what various hardware or firmware will tolerate, maybe some users succeed in using WPA2 on a WiFi 7 deployment, despite it not being allowed.

Why would anybody use WPA2 even if it is supported on a standard where all clients support WPA3 anyway. So yes just make it mandatory to use WPA3

systemcrash, 2025-01-12:
To support older devices, is one reason. But if a device supports 6ghz, then it'll more than likely support 5 and 2ghz. 

> but instead the operating mode BE (EHT) that should force the use of WPA3 or OWE.

I don't see any operating mode which mandates wpa3. Only the 6ghz band. Does your radio 2 have dual band support (including 6ghz)?

rmandrad, 2025-01-12:
yes, the mt7996 has three radios 
[image]

OpenWRT-fanboy, 2025-01-13:
@systemcrash Thanks for taking this up. I tested your new wireless.js on Linksys MX8500 and it works as expected. The 6ghz settings (Qualcomm Atheros QCN6024/9024/9074 802.11ax) now only allow WPA3 and OWE and so the radio comes up with either option.

There is another quirk. 802.11w Management Frame Protection must be set to Required or the radio won't come up. Does it make sense to turn off the other options on this band?

Interestingly, on 2ghz (Qualcomm Atheros IPQ8074 802.11ax/b/g/n) and 5ghz (Qualcomm Atheros IPQ8074 802.11ac/ax/n), 802.11w Management Frame Protection can be set to Disabled, Optional, or Required even in WPA3/OWE mode.

Somewhat related: https://github.com/openwrt/luci/issues/5394

systemcrash, 2025-01-13:
> There is another quirk. 802.11w Management Frame Protection must be set to Required or the radio won't come up. Does it make sense to turn off the other options on this band?

MFP is mandated (since it's based on 6GHz). That can go into this change. 

> Interestingly, on 2ghz (Qualcomm Atheros IPQ8074 802.11ax/b/g/n) and 5ghz (Qualcomm Atheros IPQ8074 802.11ac/ax/n), 802.11w Management Frame Protection can be set to Disabled, Optional, or Required even in WPA3/OWE mode.

MFP is not mandated in other bands, thus setting those options as so is possible. Increased security and MFP in e.g. 5GHz is allowed (undefined?), while MFP and WPA3 is mandated in 6GHz. See first link in top post.

systemcrash, 2025-01-13:
> Management Frame Protection must be set to Required or the radio won't come up

@blogic: you wrote:
just following this on the side ... mac80211.sh will override whatever uci has set for 80211w when wpa3 is set and use 2 (required)

But I cannot find this. Maybe it was removed (regression?) or changed location?

systemcrash, 2025-01-13:
Note to self: https://www.wi-fi.org/file/wpa3-specification

OpenWRT-fanboy, 2025-01-13:
> @blogic: you wrote: just following this on the side ... mac80211.sh will override whatever uci has set for 80211w when wpa3 is set and use 2 (required)
>
> But I cannot find this. Maybe it was removed [...]

This seems to me to be an odd way to do things: We'll let you configure it any way you want, but then we'll change it behind your back to fix it.

Regardless, this would presumably fully address this issue, so must not be currently implemented as described.

rmandrad, 2025-01-13:
Check 

https://github.com/rmandrad/openwrt/blob/main/package%2Fnetwork%2Fconfig%2Fwifi-scripts%2Ffiles%2Flib%2Fnetifd%2Fwireless%2Fmac80211.sh

Hostapd.sh is just one level up

OpenWRT-fanboy, 2025-01-18:
> By the sounds of things, I can go ahead and add a fix. It should help in new deployments, in any case. 24 is probably a good point to add it since we recently added WiFi 7 support (in luci).

@systemcrash Appears like rc6 and final are imminent, so if you want it in 24.10, need to merge ASAP.

rmandrad, 2025-01-19:
fyi - just noticed that iface.uc also manages security
https://git.openwrt.org/?p=openwrt/openwrt.git;a=blob;f=package/network/config/wifi-scripts/files-ucode/usr/share/ucode/wifi/iface.uc;h=710ded10e51cc8ee71d46ee9609c7ffe539ce398;hb=7482296ebf73d3c9b7f40e2f37670d337137598f

Neustradamus, 2025-07-26:
Any progress on this issue?