openwrt/luci #7554: mod-app-ddns: Absent Interface: "wan" shown by default for the Interface IP address source

adelton, 2025-01-10:
### Is there an existing issue for this?

- [X] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

1. When IP address source is changed to Interface, the Interface popup list has default selection of Absent Interface: "wan"

### Expected behaviour

1. The default selection should likely be the Alias Interface: "@wan"

### Steps to reproduce

1. In `System > Software` click `Update lists ...`
2. In `Filter` type `luci-app-ddns`
3. In the `Available` tab see the line with `luci-app-ddns`, click `Install ...`
4. Reload the page (for example with Ctrl+R)
5. A new `Services` top level menu item should appear between `System` and `Network`
6. In `Services > Dynamic DNS` click `Add new services ...`
7. Put in Name `test`, click `Create service`
8. Select the `Advanced Settings` tab
9. Change the IP address source from `Network` to `Interface`
10. Check the Interface selection and default value.

### Additional Information

```shell
{
	"kernel": "6.6.69",
	"hostname": "OpenWrt",
	"system": "Intel(R) Core(TM) i7-10610U CPU @ 1.80GHz",
	"model": "QEMU Standard PC (Q35 + ICH9, 2009)",
	"board_name": "qemu-standard-pc-q35-ich9-2009",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "24.10.0-rc5",
		"revision": "r28304-6dacba30a7",
		"target": "x86/64",
		"description": "OpenWrt 24.10.0-rc5 r28304-6dacba30a7",
		"builddate": "1736026537"
	}
}
```

### What browsers do you see the problem on?

Firefox

adelton, 2025-01-10:
I got to look at the behaviour of those selections while looking at the possibility of addressing https://github.com/openwrt/luci/issues/4229. Getting the defaults changes might not be as easy as I initially thought, so I'm filing this specific weirdness to be possible address independently from the IPv4 / IPv6 changes.

systemcrash, 2025-01-10:
I believe wan is the default since it exists on most devices by default, and @alias interfaces require (or previously have required) extra lookups to determine the L3/L2 device it refers to. I *think* the ddns-scripts take this into account.

systemcrash, 2025-01-10:
You might want to check which exists, in some descending preference order, or just skip wan entirely and mandate that the user choose an interface.

systemcrash, 2025-01-10:
(in any case, `wan` exists on my system).

adelton, 2025-01-12:
@systemcrash What would be the right idiomatic way of checking for an existence of an interface in code?

adelton, 2025-01-12:
Ah, got a reasonably working code with `network.getDevice('wan')`, shown in https://github.com/openwrt/luci/pull/7555 now.