openwrt/luci #7563: luci-app-firewall: Improper processing of multi-selects for "Allow forward..." within Zone Settings

Synchronos, 2025-01-14:
### Is there an existing issue for this?

- [X] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

Any multi-select in either the "Allow forward to destination zones:" or "Allow forward from source zones:" results in an "Expecting: valid UCI identifier" error and an inability to Save. This appears to have been recently introduced.

### Expected behaviour

The multi-select should be valid and saving should proceed as normal.

### Steps to reproduce

1. Go to Network -> Firewall
2. Find the wan zone and click the Edit button for it.
3. In allow forward from source zones, select more than one zone.
4. Hover over the box after the select has been performed.
5. Note the Expecting: valid UCI identifier
6. Note the inability to Save.

### Additional Information

```shell
{
        "kernel": "6.6.69",
        "hostname": "redacted",
        "system": "ARMv8 Processor rev 4",
        "model": "OpenWrt One",
        "board_name": "openwrt,one",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0-rc5",
                "revision": "r28304-6dacba30a7",
                "target": "mediatek/filogic",
                "description": "OpenWrt 24.10.0-rc5 r28304-6dacba30a7",
                "builddate": "1736026537"
        }
}

luci-app-firewall	25.011.06450~2520588
```

### What browsers do you see the problem on?

Chrome, Microsoft Edge

mpmc, 2025-01-14:
I get a similar issue trying to use "Any zone (forward)".

[image]

systemcrash, 2025-01-14:
Thanks for bring this to our attention. Resolved in 7046a1cac1425477196a0d49c3fc64ebfcf9a238. Evidently enforcing a datatype on the widget (in the way it was done) did not handle multiple values well. That's been removed.

You can wait a few days for an updated luci package, or replace [this file](https://github.com/openwrt/luci/blob/7046a1cac1425477196a0d49c3fc64ebfcf9a238/modules/luci-base/htdocs/luci-static/resources/tools/widgets.js) immediately. Or edit it yourself to remove as follows:
```diff
			dropdown_items: this.dropdown_size || this.size || 5,
			validate: L.bind(this.validate, this, section_id),
-			datatype: L.hasSystemFeature('firewall4') ? 'uciname' : 'and(uciname,maxlength(11))',
			create: !this.nocreate,
			create_markup: '' +

```

systemcrash, 2025-01-14:
@adelton your original commit was removed, but the result ended up being removing length restrictions elsewhere, via the check for fw4 (instead of adding the restriction in the widget). There may be other areas where the Zone widget in other places can receive a similar treatment (removing the length check via fw4 check)

adelton, 2025-01-14:
Ah, sorry about that. Would you like to reopen https://github.com/openwrt/luci/issues/7522 as that is effectively no longer resolved?

systemcrash, 2025-01-14:
The Firewall zones page was fixed not to limit the name length to 11 characters. So, problem resolved I think.

adelton, 2025-01-14:
I've now filed https://github.com/openwrt/luci/pull/7564 which adds the `list(...)` for the `this.multiple` cases.