openwrt/luci #7569: luci-mod-network: attempt to add a DNS-RR record leads to TypeError `arr.replace is not a function`

adelton, 2025-01-18:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]
[image]
[image]

### Actual behaviour

Clicking `Add` for DNS-RR does not provide a form / fields to add a new record.

Saving `uci add dhcp dnsrr` leads to TypeError `arr.replace is not a function`.

### Expected behaviour

Adding a DNS-RR record should be possible without causing a TypeError.

### Steps to reproduce

1. Get a clean slate dhcp / dnsmasq configuration by doing `cp /rom/etc/config/dhcp /etc/config/dhcp`.
2. Click `Network > DHCP and DNS > DNS Records > DNS-RR`.
3. Observe the tables says *This section contains no values yet* and there is an `Add` button.
4. Click the `Add` button.
5. Observe nothing visible happens (like fields for a new record) but in the top right corner there is `Unsaved changes: 7`.
6. Click it.
7. Observe there is a bunch of `uci del`s and one line `uci add dhcp dnsrr # =cfg04dc8e`.
8. Click `Save & Apply`.

### Additional Information

```shell
{
	"kernel": "6.6.69",
	"hostname": "OpenWrt",
	"system": "Intel(R) Core(TM) i7-10610U CPU @ 1.80GHz",
	"model": "QEMU Standard PC (Q35 + ICH9, 2009)",
	"board_name": "qemu-standard-pc-q35-ich9-2009",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "24.10.0-rc5",
		"revision": "r28304-6dacba30a7",
		"target": "x86/64",
		"description": "OpenWrt 24.10.0-rc5 r28304-6dacba30a7",
		"builddate": "1736026537"
	}
}
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell

```

dannil, 2025-01-18:
This was fixed by me in https://github.com/openwrt/luci/pull/7532 but it didn't end up in rc5, so just wait for rc6 or the stable release.

systemcrash, 2025-01-19:
Upgrading the various luci- components on the system is also an option.

closed 2025-01-19 by systemcrash