openwrt/luci #7570: LuCI: Issue with Sorting and Rendering in Static Leases

twc3000, 2025-01-18:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

In the LuCI interface under Network > DHCP and DNS > Static Leases, the sorting functionality for the fields such as hostname, MAC address, IPv4, DUID, IP-6 suffix, tag, and match tag is not working correctly. For example after adding a tag to a device, clicking on the tag column to sort does not function as expected. The code glitches and does not render the fields correctly.

Example: Sorted by Hostname ˅ - (Working Correctly):
[image]

Example: Sorted by Tag
[image]

### Actual behaviour

The fields are not rendered correctly, and the sorting by tag does not work as intended, some of the other columns do the same, Hostname works as expect with the arrow ˄ or ˅ at the end of hostname, see image of Tag column sorted.

### Expected behaviour

The entries should be correctly sorted based on the tag name, and all fields should be rendered properly without any glitches. 

### Steps to reproduce

1. Login with Chrome, Edge, or Firefox
2. Navigate to Network > DHCP and DNS > Static Leases.
3. Add a couple devices to Static Leases
4. Add a tag or tags to a devices. 
5. Attempt to sort the entries by clicking on the "Tag" column header.
6. Observe that the fields do not render correctly and the sorting does not function.

### Additional Information

```shell
{
        "kernel": "5.15.167",
        "hostname": "WRT1900ACS",
        "system": "ARMv7 Processor rev 1 (v7l)",
        "model": "Linksys WRT1900ACS",
        "board_name": "linksys,wrt1900acs",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "23.05.5",
                "revision": "r24106-10cc5fcd00",
                "target": "mvebu/cortexa9",
                "description": "OpenWrt 23.05.5 r24106-10cc5fcd00"
        }
}
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell

```

dannil, 2025-01-18:
Sorting by any column works for me on 24.10.0-rc4, so try upgrading and see if you can replicate it.

systemcrash, 2025-01-19:
@Ramon00 and I squashed a bug that I *think* affected that table (among others). The screenshot even shows the buggy "wait" animation after sorting. The fix is in 24.

twc3000, 2025-01-20:
Yes, I upgraded, to 24.10.0-rc5 and the issue doesn't exit.

closed 2025-01-20 by twc3000