openwrt/luci #7571: Feature request: Snort3 GUI

allanext, 2025-01-19:
### What would you like to see in luci?

Has a Snort3 GUI for LuCI ever been considered?

systemcrash, 2025-01-19:
Not that I know of.

efahl, 2025-01-19:
I think two things discourage most people who think about it:

1) The user community for snort on OpenWrt is tiny, it's just not that useful on CPE and requires a "big router" (x86/ARM) with lots of cpu and ram;
2) The number options that snort offers are truly enormous, so it would be pretty daunting trying to pick out which ones make sense.

Related to the latter, though, see what I did with the CLI auto-config in https://github.com/openwrt/packages/blob/master/net/snort3/files/snort.config  Note that I explicitly left the ability to inject raw Lua into the config files via that `include` option just because everyone wants something extra and different.

allanext, 2025-02-04:
I'm using OpenWRT with a cheap Linksys MX4200v1 that has ARMv8 Processor with 4 cores @ 2GHz and 364MB of RAM.

I have Snort on a Synology RT6600ax that has a Qualcomm Atheros IPQ6018 quad-core ARM 64-bit A53 1.8 GHz processor with 1 GB of RAM.

Synology has a Snort wrap/gui that calls "Threat Prevention" with 5 panels: Overview, Events, Self Defined Policies, Statistics and Settings