openwrt/luci #7573: package conflict in NIGHTLY OpenWRT

bittin, 2025-01-20:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### Actual behaviour

root@rainbowdash:~# apk add luci
ERROR: unable to select packages:
  libuci20250117-2025.01.17~fb3c2343-r1:
    conflicts: libuci20130104-2024.11.26~10f7996e-r1[libuci=2025.01.17~fb3c2343-r1]
    satisfies: world[libuci20250117] libiwinfo20230701-2024.10.20~b94f066e-r1[libuci20250117]
               netifd-2024.12.17~ea01ed41-r1[libuci20250117] odhcpd-ipv6only-2024.05.08~a2988231-r1[libuci20250117]
               procd-ujail-2024.12.22~42d39376-r1[libuci20250117] ubox-2024.04.26~85f10530-r1[libuci20250117]
               uci-2025.01.17~fb3c2343-r1[libuci20250117] ucode-mod-uci-2024.12.06~209f041f-r1[libuci20250117]
  libuci20130104-2024.11.26~10f7996e-r1:
    conflicts: libuci20250117-2025.01.17~fb3c2343-r1[libuci=2024.11.26~10f7996e-r1]
    satisfies: rpcd-2024.12.02~cc9a471c-r1[libuci20130104]

### Expected behaviour

luci installing?

### Steps to reproduce

flash OpenWRT Nightly from: https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/ 
Luci won't installl

### Additional Information

```shell
{
	"kernel": "6.6.71",
	"hostname": "rainbowdash",
	"system": "ARMv7 Processor rev 1 (v7l)",
	"model": "Linksys WRT1900ACS",
	"board_name": "linksys,wrt1900acs",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r28632-9454331b7f",
		"target": "mvebu/cortexa9",
		"description": "OpenWrt SNAPSHOT r28632-9454331b7f",
		"builddate": "1737284286"
	}
}
```

### Relevant log output

```shell

```

hnyman, 2025-01-21:
It is temporary mismatch due to recent package version upgrades, as some of those packages have ABI versioning and the two separate buildbots have not yet built all related packages according to the new compatibility specs.

Try again in 1-2 days.

closed 2025-01-21 by hnyman

bittin, 2025-01-21:
works now again

bittin, 2025-01-23:
and now it does not work again

bittin, 2025-01-23:
This is fixed now again

sateuwdie, 2025-02-01:
Today I see this

apk add luci
ERROR: unable to select packages:
  libuci20130104-2024.11.26~10f7996e-r1:
    conflicts: libuci20250120-2025.01.20~16ff0bad-r1[libuci=2024.11.26~10f7996e-r1]
    satisfies: world[libuci20130104] netifd-2024.12.17~ea01ed41-r1[libuci20130104] odhcpd-ipv6only-2024.05.08~a2988231-r1[libuci20130104] procd-ujail-2024.12.22~42d39376-r1[libuci20130104]
               ubox-2024.04.26~85f10530-r1[libuci20130104] uci-2024.11.26~10f7996e-r1[libuci20130104] ucode-mod-uci-2024.12.06~209f041f-r1[libuci20130104]
  libuci20250120-2025.01.20~16ff0bad-r1:
    conflicts: libuci20130104-2024.11.26~10f7996e-r1[libuci=2025.01.20~16ff0bad-r1]
    satisfies: libiwinfo20230701-2024.10.20~b94f066e-r1[libuci20250120] rpcd-2024.12.02~cc9a471c-r1[libuci20250120]

c3kkos, 2025-08-26:
I have this issue again on a snapshot for Mercusys MB130. Is it another mismatch issue?