openwrt/luci #7578: Feature request: Support openssh-server

zeha, 2025-01-23:
### What would you like to see in luci?

I'd like to use openssh-server as a replacement for dropbear, because of https://github.com/openwrt/openwrt/issues/7984

However there seems to be no support in luci for openssh-server. I don't know if openssh-server itself needs to add support for uci first.

I'm happy with the options luci exposes for dropbear, so if the same options could be made work for openssh-server, that'd be great.

 ❤  thank you!

zeha, 2025-01-23:
> I don't know if openssh-server itself needs to add support for uci first.

@habbie pointed out I should check the `sshd` init script about that, and it seems like it doesn't interact with uci at all.

hnyman, 2025-01-23:
Dropbear is not related to LuCI GUI.

closed 2025-01-23 by hnyman