openwrt/luci #7583: Feature request: Allow deleting DHCP lease from Luci

carefulcomputer, 2025-01-26:
### What would you like to see in luci?

Currently once a device is assigned DHCP ip address, the current value can be set as static by clicking a button. However for reverse, there is no way to remove current DHCP lease. Removing temporary lease requires ssh to router and manual deletion of row.
Please create a button (next to static lease) to allow deleting the temporary DHCP lease.

carefulcomputer, 2025-01-26:
I see issue https://github.com/openwrt/luci/issues/6979 which discusses same point.
Please just allow deleting the entry from tmp file from UI. Clients need to manage their own renewal (by restart of networking stack and openwrt/luci doesn't need to worry about it).

carefulcomputer, 2025-01-26:
I see previous issue was close. Please just provide the button to remove the entry out of dhcp lease table. Clients can manage their release/renew on their own (by restart of network stack/device)

ifsnop, 2025-04-18:
It is useful if you assign a static lease (with a different ip) to a host and need to force a refresh. Because the old lease is already in leases file, it doesn't mater if you restart the client, the old ip will be assigned until the lease expires (or you manually delete the row in the leases file).

HeyItsJono, 2025-07-17:
> It is useful if you assign a static lease (with a different ip) to a host and need to force a refresh. Because the old lease is already in leases file, it doesn't mater if you restart the client, [...]

This is exactly why i need this feature.

jezmck, 2026-10-12:
New to OpenWRT, and am stunned that this isn't already a feature.