openwrt/luci #7584: wireless: missing security protocol on multiple SSID summary

EPinci, 2025-01-26:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

When having more than one SSID on the same radio, security setup in summery is not reported.
In the capture above, the second SSID is configured with the same security as the first but that is not reported.

### Expected behaviour

Secondary SSID security configuration to be properly shown

### Steps to reproduce

1. Configure 2 SSID on the same radio with some security setting
2. Notice the security description is missing on the secondary SSID

### Additional Information

```shell
{
        "kernel": "6.6.73",
        "hostname": "Spider",
        "system": "ARMv8 Processor rev 0",
        "model": "Bananapi BPI-R4",
        "board_name": "bananapi,bpi-r4",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10-SNAPSHOT",
                "revision": "r0-1e9966a",
                "target": "mediatek/filogic",
                "description": "OpenWrt 24.10-SNAPSHOT r0-1e9966a",
                "builddate": "1737808453"
        }
}
```

### What browsers do you see the problem on?

Microsoft Edge

### Relevant log output

```shell

```

EPinci, 2025-01-26:
[image]

Same issue appears also on the LuCI dashboard.

fridtjof, 2025-01-27:
This seems to stem from `iwinfo`, which is used by luci's rpc server to query details about a wifi interface (per SSID per radio). Running `iwinfo <ifname> info` seems pretty broken for secondary SSIDs on the same radio:

```
root@gateway-wrt:/var/run# iwinfo phy0.0-ap0 info
phy0.0-ap0 ESSID: "SomeSSID"
          Access Point: <redacted, but valid>
          Mode: Master  Channel: 3 (2.422 GHz)  HT Mode: EHT20
          Center Channel 1: 3 2: unknown
          Tx-Power: 6 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -82 dBm
          Bit Rate: unknown
          Encryption: mixed WPA2/WPA3 PSK/SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
root@gateway-wrt:/var/run# iwinfo phy0.0-ap1 info
phy0.0-ap1 ESSID: unknown
          Access Point: <redacted, but valid>
          Mode: Master  Channel: 3 (2.422 GHz)  HT Mode: EHT20
          Center Channel 1: 3 2: unknown
          Tx-Power: 6 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -82 dBm
          Bit Rate: unknown
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11ac/ax/b/be/g/n
          Hardware: 14C3:7990 14C3:6639 [MediaTek MT7996E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0
```

No idea why the ESSID is not showing up for -ap1, but it seems unrelated (different code path).

The way iwinfo gets encryption information is (if using hostapd) by parsing hostapd's config file (or asking wpa_supplicant but I think that isn't relevant for AP operation?), so I suspect something is subtly wrong there?

fridtjof, 2025-01-28:
Found it.
Iwinfo tries figuring out the radio index (which it needs for the hostapd config file name. what a mess) from the interface name through netlink, which should work! Netlink returns a radio mask, which iwinfo parses to find out one (!) radio ID. I think it's a mask for MLO so you can have an interface with multiple radios involved?

Now MLO is not yet relevant here, I think, so there shouldn't be any interface with more than one radio in use (and in this case, it should obviously be radio 0 in both cases).

```
Interface phy0.0-ap0
        ifindex 18
        wdev 0x8
        addr <redacted>
        ssid <SSID 1>
        type AP
        wiphy 0
        channel 9 (2452 MHz), width: 20 MHz, center1: 2452 MHz
        txpower 6.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       0       0       0       0       0       0               0
        Radios: **0**
root@gateway-wrt:/etc/apk# iw dev phy0.0-ap1 info
Interface phy0.0-ap1
        ifindex 19
        wdev 0x9
        addr <redacted>
        ssid <SSID 2>
        type AP
        wiphy 0
        channel 9 (2452 MHz), width: 20 MHz, center1: 2452 MHz
        txpower 6.00 dBm
        multicast TXQ:
                qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                0       0       0       0       0       0       0       0               0
        Radios: **0 1 2**
```

Note the bolded parts - the second SSID interface uses all three radios for some reason??????? That leads iwinfo to just pick the highest bit in the radio mask (which corresponds to radio 2), read the wrong hostapd config, where it fails to find the actual interface name it's looking for.

EPinci, 2025-01-28:
I can confirm I see the same output in my BPI-r4.
Secondary SSID has all the three radios where it should have only one.

EPinci, 2025-01-28:
Is definitely MLO associated: the same issue is not showing on my Beryl (wifi6) running the same openwrt build.

EPinci, 2025-05-27:
@nbd168 can you help with `iw` and `iwinfo` misbehaving with whipy?