openwrt/luci #7589: luci_app_acme: missing support for dns_dnsexit - solution suggested

tboege, 2025-01-28:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

`cat /etc/config/acme

config cert '_redacted_'
        option enabled '1'
        list domains '_redacted_'
        option update_uhttpd '1'
        option validation_method 'dns'
        option dns 'dns_dnsexit'
        list credentials 'DNSEXIT_API_KEY=_redacted_'
        list credentials 'DNSEXIT_AUTH_USER=_redacted_'
        list credentials 'DNSEXIT_AUTH_PASS=_redacted_'
        option dalias '_acme-challenge-_redacted_'
        option days '170'
        option use_acme_server '1'
        option staging '0'
        option key_type 'ec256'
		
		`

### Actual behaviour

when /etc/config/acme has
`   option dns 'dns_dnsexit'`
the field DNS api is empty (you cannot select DNSExit.com)

### Expected behaviour

You should be able to select DNSExit.com

The solution is simple:
just add to 
/www/luci-static/resources/view/acme.js
`o.value('dns_dnsexit', 'DNSExit.com');`
between the lines
`o.value('dns_dgon', 'DigitalOcean.com');
o.value('dns_dnshome', 'dnsHome.de');`

### Steps to reproduce

1. Go to Services -> ACME Certificates -> Edit example_wildcard -> DNS Challenge Validation and open DNS API
2. You cannot find an entry for DNSExit.com

### Additional Information

```shell
{
        "kernel": "6.6.67",
        "hostname": "nelson",
        "system": "ARMv7 Processor rev 0 (v7l)",
        "model": "TP-Link Archer C2600",
        "board_name": "tplink,c2600",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "24.10.0-rc4",
                "revision": "r28211-d55754ce0d",
                "target": "ipq806x/generic",
                "description": "OpenWrt 24.10.0-rc4 r28211-d55754ce0d",
                "builddate": "1734915335"
        }
}
```

### Relevant log output

```shell

```

tboege, 2025-01-28:
A better solution is suggested in PR https://github.com/openwrt/luci/pull/7515

systemcrash, 2025-01-29:
@tohojo