openwrt/luci #7591: luci-app-firewall: cannot choose `Any zone` in `Output zone` field

this-username-has-been-taken, 2025-01-29:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

Hello!

I have just build and installed a fresh firmware from the `master`/`main` branch and faced with an issue: I cannot add or modify a firewall rule and set `Output zone` to `Any zone`: LuCI shows an error: `Expecting: valid UCI identifier`. Please refer to the attached screenshot for more details.

It is possible to modify the rule via editing the config file, but that's not very convenient.

I have searched the similar issues and found this: https://github.com/openwrt/luci/issues/7563
I believe my issue might be connected with it.

[image]

### Actual behaviour

When adding or modifying a firewall rule it is impossible to set `Output zone` field to `Any zone` value. LuCI show an error: `Expecting: valid UCI identifier`.

### Expected behaviour

When adding or modifying a firewall rule it is possible to set `Output zone` field to `Any zone` value. LuCI accepts the value and saves it in the config file without any errors.

### Steps to reproduce

1. Obtain a freshly build (preferably based on `master`/`main` branch) firmware and install it.
2. Log in into LuCI.
3. Navigate to Network -> Firewall -> Traffic rules.
4. Add a new rule and set `Output zone` field to `Any zone`.
5. Observe the issue: LuCI highlights the value with red and shows `Expecting: valid UCI identifier` error. It is impossible to save the rule

### Additional Information

```shell
{
        "kernel": "6.6.74",
        "hostname": "router",
        "system": "ARMv8 Processor rev 0",
        "model": "Bananapi BPI-R4",
        "board_name": "bananapi,bpi-r4",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r28724-807074309d",
                "target": "mediatek/filogic",
                "description": "OpenWrt SNAPSHOT r28724-807074309d",
                "builddate": "1738094344"
        }
}
```

### What browsers do you see the problem on?

Firefox

### Relevant log output

```shell

```

systemcrash, 2025-01-29:
Hmm. Evidently there remain edge cases for the widget. My guess is that the name with a space in it is the trigger here. 

Any suggestions @adelton ?

systemcrash, 2025-01-29:
@this-username-has-been-taken I wonder whether you can test whether the following works:
```js
			datatype: L.hasSystemFeature('firewall4')
				? ( this.multiple ? 'list(string)' : 'string' )
				: this.multiple ? 'list(and(string,maxlength(11)))' : 'and(string,maxlength(11))',

```

It's not ideal, but at least fixes this edge-case, I think (assuming `string` permits the space in this 'name').

this-username-has-been-taken, 2025-01-29:
@systemcrash , just checked: yes, that worked!
With that code it is now possible to choose `Any zone` as well as "real" zones:

[image]

[image]

systemcrash, 2025-01-29:
> But I don't remember if I was able to delete a chosen `Output zone` or not before

Delete?

adelton, 2025-01-29:
@systemcrash Is it expected that when I install OpenWrt 24.10.0-rc7 (r28417-daef29c75d) and upgrade all upgradable packages, giving me
```
# opkg list-installed | grep luci | sort
liblucihttp-ucode - 2023.03.15~9b5b683f-r1
liblucihttp0 - 2023.03.15~9b5b683f-r1
luci - 25.027.83426~170375e
luci-app-firewall - 25.027.83426~170375e
luci-app-package-manager - 25.027.83426~170375e
luci-base - 25.027.83426~170375e
luci-light - 25.027.83426~170375e
luci-mod-admin-full - 25.027.83426~170375e
luci-mod-network - 25.027.83426~170375e
luci-mod-status - 25.027.83426~170375e
luci-mod-system - 25.027.83426~170375e
luci-proto-ipv6 - 25.027.83426~170375e
luci-proto-ppp - 25.027.83426~170375e
luci-ssl - 25.027.83426~170375e
luci-theme-bootstrap - 25.027.83426~170375e
rpcd-mod-luci - 20240305-r1
```
that I don't reproduce that issue?

In fact,
```
root@OpenWrt:~# grep firewall4 /www/luci-static/resources/tools/widgets.js
```
does not find anything.

Does it mean we don't have a package which would contain https://github.com/openwrt/luci/pull/7564? I'm trying to establish a baseline for a reproducer (and a fix).

this-username-has-been-taken, 2025-01-29:
> Delete?

I meant that when you open a new rule dialog the `Output zone` is empty:

[image]

If you save a rule like this (with an empty value - not choosing anything) - the zone value will be set to `this device`.
After you choose something, you won't be able to make it empty again, i.e. it won't be possible to set `this device` value

adelton, 2025-01-29:
@this-username-has-been-taken: Could you check the code in https://github.com/openwrt/luci/pull/7592?

adelton, 2025-01-29:
As for removing the selection completely after either a zone or Any zone has been selected: when I modify the code to drop the `datatype` completely, bringing the code to the state before a075566dbb5147a4311cd24f46a173222c9172cf, I'm not able to completely clear the selection either. That would suggest that it was not possible to clear the selection even with the previous versions (23.05.5) so that behaviour is not a regression.

this-username-has-been-taken, 2025-01-29:
@adelton , I have just checked: the code from https://github.com/openwrt/luci/pull/7592 works!

I'd suggest putting "else" part under parentheses too to make it look uniform, but that's not critical:
```
datatype: L.hasSystemFeature('firewall4')
				? ( this.multiple ? 'list(or(uciname,"*"))' : 'or(uciname,"*")' )
				: ( this.multiple ? 'list(or(and(uciname,maxlength(11)),"*"))' : 'or(and(uciname,maxlength(11)),"*")' ),
```

As per `this device` value - it works either! You just have to choose anything besides `Device (output)` in the `Source zone` field - and viola - you can now choose `Device (input)` in the `Destination zone` field. I forgot about that.

systemcrash, 2025-01-29:
> [@systemcrash](https://github.com/systemcrash) Is it expected that when I install OpenWrt 24.10.0-rc7 (r28417-daef29c75d) and upgrade all upgradable packages, giving me

Your PR was merged to main (snapshots) but not to 24. I figured some edge-cases might still be lingering.

closed 2025-01-30 by systemcrash

this-username-has-been-taken, 2025-01-31:
Thank you very much for a quick fix!!! :)