openwrt/luci #7594: luci-mod-dashboard: RSSI bigger than 100% layout blow up

webysther, 2025-01-30:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

### Actual behaviour

The bar blow up when the RSSI is bigger than 100%

### Expected behaviour

The get filled if RSSI >= 100%

### Steps to reproduce

N/A

### Additional Information

```shell
{
"kernel": "5.15.167",
"hostname": "plsi11",
"system": "MediaTek MT7621 ver:1 eco:3",
"model": "Xiaomi Mi Router 4A Gigabit Edition v2",
"board_name": "xiaomi,mi-router-4a-gigabit-v2",
"rootfs_type": "squashfs",
"release": {
"distribution": "OpenWrt",
"version": "23.05.5",
"revision": "r24106-10cc5fcd00",
"target": "ramips/mt7621",
"description": "OpenWrt 23.05.5 r24106-10cc5fcd00"
}
}
```

### Relevant log output

```shell

```

systemcrash, 2025-01-30:
I think those RSSI values are just misinterpreted values, somehow SSID was skipped. So the table values are misaligned. Do your SSID have unusual characters in the names?

webysther, 2025-01-30:
> I think those RSSI values are just misinterpreted values, somehow SSID was skipped. So the table values are misaligned. Do your SSID have unusual characters in the names?

SSID is working is the align of the table make you believe is empty. The second red column is the SSID. The RSSI come from the server with the value, if check on front to round to 100% when is bigger I think this fixes the visual part of the problem. Another point is discovery way the data is bigger than 100%.