openwrt/luci #7595: luci-mod-dashboard: Duplicated table broke align

webysther, 2025-01-30:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

[image]

[image]

### Actual behaviour

Double table broke the data and column title align

### Expected behaviour

One table to naturally keep the align

### Steps to reproduce

N/A

### Additional Information

```shell
{
"kernel": "5.15.167",
"hostname": "plsi11",
"system": "MediaTek MT7621 ver:1 eco:3",
"model": "Xiaomi Mi Router 4A Gigabit Edition v2",
"board_name": "xiaomi,mi-router-4a-gigabit-v2",
"rootfs_type": "squashfs",
"release": {
"distribution": "OpenWrt",
"version": "23.05.5",
"revision": "r24106-10cc5fcd00",
"target": "ramips/mt7621",
"description": "OpenWrt 23.05.5 r24106-10cc5fcd00"
}
}
```

### What browsers do you see the problem on?

Chrome, Microsoft Edge

### Relevant log output

```shell

```

adelton, 2025-01-30:
I believe the situation was significantly improved (if not completely fixed) via https://github.com/openwrt/luci/pull/7550. When I upgrade to luci-mod-dashboard 25.027.83426~170375e, the table looks reasonable.

webysther, 2025-02-12:
Fixed on 24.10

closed 2025-02-12 by webysther