openwrt/luci #7605: luci-app-keepalived: Unable to find this package in ‘make menuconfig’

NNET-ELi, 2025-02-05:
### Is there an existing issue for this?

- [x] I have searched among all existing issues (including closed issues)

### screenshots or captures

I can't find this package in make menuconfig. Can you tell me where to choose? Thanks.

version: Official 24.10.0
url: https://github.com/openwrt/openwrt/archive/refs/tags/v24.10.0.tar.gz

[image]
[image]

### Actual behaviour

Find this package in make menuconfig

### Expected behaviour

Can find and enable

### Steps to reproduce

Find this package in make menuconfig

[image]

### Additional Information

```shell
{
	"kernel": "6.6.73",
	"hostname": "OpenWrt",
	"system": "ARMv8 Processor rev 4",
	"model": "Cudy TR3000 v1",
	"board_name": "cudy,tr3000-v1",
	"rootfs_type": "squashfs",
	"release": {
		"distribution": "OpenWrt",
		"version": "24.10.0",
		"revision": "r28427-6df0e3d02a",
		"target": "mediatek/filogic",
		"description": "OpenWrt 24.10.0 r28427-6df0e3d02a",
		"builddate": "1738624177"
	}
}
```

### Relevant log output

```shell

```

feckert, 2025-02-12:
It works in my setup.
But I can see in your [posted picture](https://private-user-images.githubusercontent.com/92837717/410058280-0e3b21de-78d7-4e73-8c53-a0d7ce571563.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkzNzcyMjYsIm5iZiI6MTczOTM3NjkyNiwicGF0aCI6Ii85MjgzNzcxNy80MTAwNTgyODAtMGUzYjIxZGUtNzhkNy00ZTczLThjNTMtYTBkN2NlNTcxNTYzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjEyVDE2MTUyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI5MGZhMDY5MDk0M2UxMjQ3ZGIzMTU3ZjExM2Y1MWEzZmVmZmYzNDlmYWZkZDE4NmViMmYxMmNhOWI0M2M3NTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.pnCNYovj-5JKH9FNNtoTe35pFQGijBuEJZnisAsWX0A) that there is a dependency to `PACKAGE_coreutils`. If this is **not** enabled, then you can not select `luci-app-keepalived`.

hnyman, 2025-02-13:
> there is a dependency to PACKAGE_coreutils

That dependency is strange as the LuCI app Makefile is straightforward selection of one specific coreutils module. 
Might be that the individual coreutils apps depend on the main coreutils umbrella being selected. Then the LuCI app should include also a `+coreutils` dependency.
Interesting case.

feckert, 2025-02-13:
It is because of the following line in [luci-app-keepalived](https://github.com/openwrt/luci/blob/master/applications/luci-app-keepalived/Makefile#L12).
The dependency is set to `keepalived-sync`. This packages has a [dependency](https://github.com/openwrt/packages/blob/master/net/keepalived/Makefile#L287) on `coreutils-timeout`.

As long as you do not activate the `coreutils` package so the buildsystem can select `coreuitls-timeout`, the dependency is not fulfilled and `luci-app-keepalived` can not be enabled

NNET-ELi, 2025-02-13:
@feckert @hnyman I have already selected this “coreutils” package before. But I still can't find it
When I added the keepalived-sync package, I was able to find it, Thank you all for your help.

[image]

[image]

feckert, 2025-02-13:
I would say this is solved

closed 2025-02-13 by feckert

hnyman, 2025-03-15:
I created the packages PR to actually solve this:  change the individual apps' dependency to empty virtual coreutils main package to be selective.
https://github.com/openwrt/packages/pull/26147