openwrt/luci #7627: ipv6 setting for wan6 also show in wan edit in luci

peterb99, 2025-02-13:
ipv6 parameters for wan6 are also shown in lan interface under advanced in LUCI
This is very confusing, can a distinction made between ipv4 and ipv6 in the luci setup windows and only show where appropriate.
Also missing a loglevel option in wan and wan6 for making dhcp request quiet in the syslog when wan/wan6 interface is using DHCP protocol.

Current OpenWRT version used:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.0'
DISTRIB_REVISION='r28427-6df0e3d02a'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 24.10.0 r28427-6df0e3d02a'
```

feckert, 2025-03-04:
Please take a few screenshots and describe what exactly doesn't suit you.
It's a difficult to understand with words.

peterb99, 2025-03-08:
wan6
[image]

[image]
wan

[image]

[image]

And then under dhcp in WAN there are more ipv6 setting.

[image]

So should all ipv6 settings not be in wan6 and under dhcp server/ipv6