openwrt/luci #7637: Set Static IP button - odhcpd use "option mac" but dnsmasq use "list mac" causing issues

skramstad, 2025-02-18:
### How can we otherwise help?

Hi.

I'm using **unbound** + **odhcpd** and when I use the "Set Static" button on the page `cgi-bin/luci/` 
it will add the clients **MAC** = `uci add_list dhcp.@host[-1].mac='3C:XX:XX:XX:XX:47'`.
And that will add the client in `/etc/config/dhcp` and `list mac '3C:XX:XX:XX:XX:47'`. This is correct and works great for **dnsmasq**. 

However, when you use **odhcpd** the MAC needs to be `option mac '3C:XX:XX:XX:XX:47'` to work.

Is there any way to check if **dnsmasq** or **odhcpd** is running on the system, and the add the **MAC** with the correct option? 

`list mac` for **dnsmasq** 
`option mac` for **odhcpd**

Thanks.

jow-, 2025-02-18:
No, this should be fixed on the odhcpd side since it hijacks the same configuration. LuCI should not differentiate.

skramstad, 2025-02-18:
Alright @jow- . I'm closing this and have a look at odhcpd.

closed 2025-02-18 by skramstad